coder-mike/microvium #1: console.log

pitust, 2020-05-13:
https://github.com/coder-mike/microvium/blob/master/doc/getting-started.md says console.log and it doesnt work. Did you mean log, which does work?

coder-mike, 2020-05-13:
Welcome! I think you may be the first person to try Microvium out, and certainly the first to contribute an issue!

I think the issue you're seeing is that you're probably using the latest (pre)release on npm, which was released before I had implemented support for object and properties like `console.log`. Things have progressed since then I changed it to `console.log` to make it more familiar to JS developers. I'll run through the getting-started guide again on my machine just to make sure it all works, and re-publish another (pre)release to npm.

Do you mind me asking, how did you hear about Microvium, since I haven't done any publicizing yet? And what are you considering using it for?

coder-mike, 2020-05-14:
@pitust See if (pre)release 0.0.8 fixes your issue. 

I also added a [regression test suite](https://github.com/coder-mike/microvium/blob/master/test/getting-started/getting-started.test.ts) which programmatically executes the code in the getting-started.md file to help it from getting out of sync again in future.

pitust, 2020-05-21:
Embedding should be made easier IMO, if I want to embed, say, wasm2c'ed
code (which lets you run Typescript via AssemblyScript),  i need to
implement (when writing with OSDev-level STL mind you): calloc and exports
and call one function. There is no "resolveImport" which by the way, can be
auto-generated, all arguments are normal C arguments (unless the language
used has them). On the flip side, structure deserialization in wasm2c is a
pain, because NOTHING lies on the stack, and every pointer is a pointer to
a different address space.

coder-mike, 2020-05-21:
> There is no "resolveImport"

I'm not familiar with wasm2c. If there's nothing analogous to `resolveImport`, then how do you call a C function TypeScript code? Or, how do you import a function implemented by the host?

pitust, 2020-05-25:
You import them in TS like this:
```typescript
declare function importedFunc(): void;
```
Then, compiled to wasm, it will be a WASM import, which wasm2c will convert into a (mangled) name.
You then create a variable with that name, assign the function poitnter to it, and it (hopefully) will work.

pitust, 2020-05-25:
Fixed

closed 2020-05-25 by pitust