coder-mike/microvium #2: colors and @babel/types

pitust, 2020-05-25:
These are dependencies, however they are not included by default in package.json for some reason.

coder-mike, 2020-05-25:
I can see them listed in the `devDependencies`. Certainly, I should move `colors` to the dependencies, but `@babel/types` shouldn't be a runtime dependency since the package entry points are JavaScript not TypeScript. Let me know if you disagree.

pitust, 2020-05-26:
Umm... When I got microvium off npm it crashed until i got `@babel/types`.

coder-mike, 2020-05-26:
Ok, I'm able to reproduce the same issue with microvium as a library. Perhaps, for some strange reason, `@babel/types` is more than just types. Anyway, I'll include it in the dependencies and the problem will go away. Thanks for uncovering this.

closed 2020-05-26 by coder-mike

pitust, 2020-05-26:
I think i had propblems with both; but for disk-space reasons i use PNPM instead of NPM. So perhaps one of your dependencies had it but PNPM didn't give it to you because it wasn't in your node_modules

commit bb53bec7 (2020-05-25): Fix #002 Moved `colors` package from `devDependencies` to `dependencies`
commit 409491be (2020-05-26): Fix #002 Moved @babel/types to dependencies instead of devDependencies