coder-mike/microvium #3: An interal compiler error

pitust, 2020-05-25:
```
Error: Internal compile error: Jumping from stack depth of 3 to block with stack depth of 1
```
```

let globalA;
let globalB = 42;
function ramGet(addr) {
    if (addr === 0) return 0x01;
    if (addr === 1) return 0x00;
    if (addr === 2) return 0x02;
    if (addr === 3) return 0x00;
    print('Access violation: 0x' + addr.toString(16));
    return 0xff;
}
let pc = 0;
function foo() {
    let ins = ramGet(pc);
    // blink_on
    if (ins === 0x01) {
        print('Lamp ON');
    }
    if (ins === 0x00) {
        print('Lamp OFF');
    }
    if (ins === 0x02) {
        pc = (ramGet(pc + 1));
    } else {
        pc = (pc + 1);
    }
}
print(foo());
```
Well, i don't think that is expected.
This is the IR (when i forced it to compile by pulling out the assertion, forgive me the TERRIBLE variable making or not-making):
```

xvf.js:#entry(param0) {
    entry:
       LoadArg param0
       StoreGlobal xvf.js:exports
       Literal undefined
       StoreGlobal xvf.js:globalA
       Literal 42
       StoreGlobal xvf.js:globalB
       Literal 0
       StoreGlobal xvf.js:pc
       LoadGlobal global:print
       Literal undefined
       LoadGlobal xvf.js:foo
       Literal undefined
       Call 1
       Call 2
       Pop 1
       var0$5 = Literal undefined
       Return var0$5
}
xvf.js:ramGet(param0, param1) {
    entry:
       param1 = LoadArg param1
       LoadVar param1
       Literal 0
       BinOp ===
       var0$7 = Branch block1 block2
    block1:
       var1$7 = Literal 1
       Return var1$7
    block2:
       var1$7 = LoadVar var0$7
       Literal 1
       BinOp ===
       var0$8 = Branch block3 block4
    block3:
       var1$9 = Literal 0
       Return var1$9
    block4:
       var1$9 = LoadVar var0$8
       Literal 2
       BinOp ===
       var0$9 = Branch block5 block6
    block5:
       var1$11 = Literal 2
       Return var1$11
    block6:
       var1$11 = LoadVar var0$9
       Literal 3
       BinOp ===
       var0$10 = Branch block7 block8
    block7:
       var1$13 = Literal 0
       Return var1$13
    block8:
       var1$13 = LoadGlobal global:print
       Literal undefined
       Literal "Access violation: 0x"
       var4$0 = LoadVar var0$10
       Dup 
       Literal "toString"
       ObjectGet 
       LoadVar var4$0
       Literal 16
       Call 2
       var4$0 = StoreVar var4$0
       BinOp +
       var1$13 = Call 2
       var0$10 = Pop 1
       var1$14 = Literal 255
       Return var1$14
}
xvf.js:foo(param0) {
    entry:
       var0$11 = LoadGlobal xvf.js:ramGet
       Literal undefined
       LoadGlobal xvf.js:pc
       var0$11 = Call 2
       LoadVar var0$11
       Literal 1
       BinOp ===
       var0$12 = Branch block1 block2
    block1:
       LoadGlobal global:print
       Literal undefined
       Literal "Lamp ON"
       Call 2
       var0$12 = Pop 1
       ; Fallthrough
    block2:
       LoadVar var0$12
       Literal 0
       BinOp ===
       var0$13 = Branch block3 block4
    block3:
       LoadGlobal global:print
       Literal undefined
       Literal "Lamp OFF"
       Call 2
       var0$13 = Pop 1
       ; Fallthrough
    block4:
       LoadVar var0$13
       Literal 2
       BinOp ===
       Branch block5 block6
    block5:
       var1$25 = LoadGlobal xvf.js:ramGet
       Literal undefined
       LoadGlobal xvf.js:pc
       Literal 1
       BinOp +
       var1$25 = Call 2
       LoadVar var1$25
       var1$25 = StoreGlobal xvf.js:pc
       Pop 1
       Jump block7
    block6:
       LoadGlobal xvf.js:pc
       Literal 1
       var1$28 = BinOp +
       LoadVar var1$28
       var1$28 = StoreGlobal xvf.js:pc
       Pop 1
       ; Fallthrough
    block7:
       var1$29 = Literal undefined
       Return var1$29
}
```
Oddly enough, these appear at jumps to block 4, 6 and 8. You do not see them because my IR dumper is smart smart and stops dumping when it hits a `Return`. Obviously.
Uncommenting that gives this IR:
```

xvf.js:#entry(param0) {
    entry:
       LoadArg param0
       StoreGlobal xvf.js:exports
       Literal undefined
       StoreGlobal xvf.js:globalA
       Literal 42
       StoreGlobal xvf.js:globalB
       Literal 0
       StoreGlobal xvf.js:pc
       LoadGlobal global:print
       Literal undefined
       LoadGlobal xvf.js:foo
       Literal undefined
       Call 1
       Call 2
       Pop 1
       var0$5 = Literal undefined
       Return var0$5
       Return 
}
xvf.js:ramGet(param0, param1) {
    entry:
       param1 = LoadArg param1
       LoadVar param1
       Literal 0
       BinOp ===
       var0$7 = Branch block1 block2
    block1:
       var1$7 = Literal 1
       Return var1$7
       Return 
       ; Fallthrough
    block2:
       var1$7 = LoadVar var0$7
       Literal 1
       BinOp ===
       var0$8 = Branch block3 block4
    block3:
       var1$9 = Literal 0
       Return var1$9
       Return 
       ; Fallthrough
    block4:
       var1$9 = LoadVar var0$8
       Literal 2
       BinOp ===
       var0$9 = Branch block5 block6
    block5:
       var1$11 = Literal 2
       Return var1$11
       Return 
       ; Fallthrough
    block6:
       var1$11 = LoadVar var0$9
       Literal 3
       BinOp ===
       var0$10 = Branch block7 block8
    block7:
       var1$13 = Literal 0
       Return var1$13
       Return 
       ; Fallthrough
    block8:
       var1$13 = LoadGlobal global:print
       Literal undefined
       Literal "Access violation: 0x"
       var4$0 = LoadVar var0$10
       Dup 
       Literal "toString"
       ObjectGet 
       LoadVar var4$0
       Literal 16
       Call 2
       var4$0 = StoreVar var4$0
       BinOp +
       var1$13 = Call 2
       var0$10 = Pop 1
       var1$14 = Literal 255
       Return var1$14
       Return 
       var3$42 = Literal undefined
       Return var3$42
       Return 
}
xvf.js:foo(param0) {
    entry:
       var0$11 = LoadGlobal xvf.js:ramGet
       Literal undefined
       LoadGlobal xvf.js:pc
       var0$11 = Call 2
       LoadVar var0$11
       Literal 1
       BinOp ===
       var0$12 = Branch block1 block2
    block1:
       LoadGlobal global:print
       Literal undefined
       Literal "Lamp ON"
       Call 2
       var0$12 = Pop 1
       ; Fallthrough
    block2:
       LoadVar var0$12
       Literal 0
       BinOp ===
       var0$13 = Branch block3 block4
    block3:
       LoadGlobal global:print
       Literal undefined
       Literal "Lamp OFF"
       Call 2
       var0$13 = Pop 1
       ; Fallthrough
    block4:
       LoadVar var0$13
       Literal 2
       BinOp ===
       Branch block5 block6
    block5:
       var1$25 = LoadGlobal xvf.js:ramGet
       Literal undefined
       LoadGlobal xvf.js:pc
       Literal 1
       BinOp +
       var1$25 = Call 2
       LoadVar var1$25
       var1$25 = StoreGlobal xvf.js:pc
       Pop 1
       Jump block7
    block6:
       LoadGlobal xvf.js:pc
       Literal 1
       var1$28 = BinOp +
       LoadVar var1$28
       var1$28 = StoreGlobal xvf.js:pc
       Pop 1
       ; Fallthrough
    block7:
       var1$29 = Literal undefined
       Return var1$29
       Return 
}
```
The thing is, none of these jumps appear. The reason again is because my tool is smart smart and figures out it is a fallthrough. So it just adds a comment `; Fallthrough`. And calls it a day.
Solution: Stop parsing on a `Return`. Actually, stop doing anything then. Even parsing the code. It is dead code. Why bother?

coder-mike, 2020-05-26:
Great, thanks for highlighting this bug!

Ok, the stack depth issue should be fixed now on `master`. Running your example code for me yields:

[image]

The only thing I changed to run this is that I changed `print` to `console.log`.

Let me know if this fixes your issue.

coder-mike, 2020-05-26:
(the `undefined` printout on the terminal is because it's doing `console.log(foo())` when `foo` doesn't return a value)

coder-mike, 2020-05-26:
> Stop parsing a block on a Return. Actually, stop doing anything then. Even parsing the code. It is dead code. Why bother?

The following Microvium source code should not run:

```js
// script.mvms
foo();
function foo() {
  console.log('foo');
  return;
  This line is unreachable and syntactically invalid, but definitely *should* give an error
}
```

Running in node and `microvium` both give syntax errors (albeit worded differently).

[image]

I'm not sure how it would even make sense to stop parsing at a runtime error.

pitust, 2020-05-26:
Do not stop parsing, but stop the codegen. I don't think it is necessary to codegen dead code, that is all. But it is fixed, so i'm closing.

closed 2020-05-26 by pitust