coder-mike/microvium #4: Error: Internal compiler error: Stack unbalanced

pitust, 2020-05-26:
```javascript
// A minimal runtime header for this thing.
let p_console = {}
p_console.val = {}
p_console.val.p_log = {}
p_console.val.p_log.val = {}
p_console.val.p_log.val.invoke = doLogging;
function doLogging(_p1, p2, p3) {
    console.log(p2, p3);
}
let ums = create_d___usermode_start();
ums.invoke(ums);
function invoke_d_v_0(self, arg0) {
    return v_0(self.p_gh, self.p_g, self.p_a, self.p_b, self.p_c, arg0);
}
function create_d_v_0() {
    return {
        invoke: invoke_d_v_0,

        p_gh: {},
        p_g: {},
        p_a: {},
        p_b: {},
        p_c: {},
    }
}

function invoke_d___usermode_start(self, ) {
    return __usermode_start(self.p_gh, self.p_g, self.p_a, self.p_b, self.p_c);
}
function create_d___usermode_start() {
    return {
        invoke: invoke_d___usermode_start,

        p_gh: {},
        p_g: {},
        p_a: {},
        p_b: {},
        p_c: {},
    }
}

const BR_v_6 = 1;
const BR_v_7 = 2;
const BR_v_13 = 3;

function v_0(p_gh, p_g, p_a, p_b, p_c, p_af) {
    var v_1 = {};
    var v_2 = {};
    var curbr = 0;
    while (true) {
        if (curbr === 0) {
                v_1.val = (7) + (p_af.val);
                v_2.val = (v_1.val) + (p_gh.val);
                return v_2.val;
                return;
        }
    }
}
function __usermode_start(p_gh, p_g, p_a, p_b, p_c) {
    var v_3 = {};
    var v_4 = {};
    var v_5 = {};
    var v_6 = {};
    var v_7 = {};
    var v_8 = {};
    var v_9 = {};
    var v_10 = {};
    var v_11 = {};
    var vcp_v_11 = {};
    var v_12 = {};
    var v_13 = {};
    var curbr = 0;
    while (true) {
        if (curbr === 0) {
                v_3.val = -(1);
                p_gh.val = (v_3.val);
                p_g.val = create_d_v_0();
                p_g.val.p_gh = p_gh;
                p_g.val.p_g = p_g;
                p_g.val.p_a = p_a;
                p_g.val.p_b = p_b;
                v_4.val = -(2);
                v_5.val = p_g.val.invoke(p_g.val, { val: (v_4.val) });
                p_a.val = (v_5.val);
                curbr = BR_v_7;
            }
if (curbr === BR_v_6) {
                p_b.val = (p_a.val);
                p_c.val = ("ABCD");
                v_8.val = (p_console.val.p_log.val);
                v_9.val = v_8.val.invoke(v_8.val, { val: ("Hello, usIR!") }, { val: (p_c.val) });
                v_10.val = (p_b.val) - (1);
                vcp_v_11.val = (v_10.val);
                p_a.val = (vcp_v_11.val);
            }
if (curbr === BR_v_7) {
                v_12.val = (p_a.val) > (0);
                if ((v_12).val) { curbr = BR_v_6; }
            }
if (curbr === BR_v_13) {
                return;
        }
    }
}
```
Causes an error:
```

Error: Internal compiler error: Stack unbalanced
    at Object.unexpected (/home/pitu/.nvm/versions/node/v13.8.0/pnpm-global/3/node_modules/.pnpm/registry.npmjs.org/microvium/0.0.8/node_modules/microvium/dist/lib/utils.js:27:11)
    at Object.endScope (/home/pitu/.nvm/versions/node/v13.8.0/pnpm-global/3/node_modules/.pnpm/registry.npmjs.org/microvium/0.0.8/node_modules/microvium/dist/lib/src-to-il.js:1210:32)
    at compileBlockStatement (/home/pitu/.nvm/versions/node/v13.8.0/pnpm-global/3/node_modules/.pnpm/registry.npmjs.org/microvium/0.0.8/node_modules/microvium/dist/lib/src-to-il.js:458:11)
    at compileStatement (/home/pitu/.nvm/versions/node/v13.8.0/pnpm-global/3/node_modules/.pnpm/registry.npmjs.org/microvium/0.0.8/node_modules/microvium/dist/lib/src-to-il.js:657:39)
    at compileIfStatement (/home/pitu/.nvm/versions/node/v13.8.0/pnpm-global/3/node_modules/.pnpm/registry.npmjs.org/microvium/0.0.8/node_modules/microvium/dist/lib/src-to-il.js:480:9)
    at compileStatement (/home/pitu/.nvm/versions/node/v13.8.0/pnpm-global/3/node_modules/.pnpm/registry.npmjs.org/microvium/0.0.8/node_modules/microvium/dist/lib/src-to-il.js:656:36)
    at compileBlockStatement (/home/pitu/.nvm/versions/node/v13.8.0/pnpm-global/3/node_modules/.pnpm/registry.npmjs.org/microvium/0.0.8/node_modules/microvium/dist/lib/src-to-il.js:456:9)
    at compileStatement (/home/pitu/.nvm/versions/node/v13.8.0/pnpm-global/3/node_modules/.pnpm/registry.npmjs.org/microvium/0.0.8/node_modules/microvium/dist/lib/src-to-il.js:657:39)
    at compileWhileStatement (/home/pitu/.nvm/versions/node/v13.8.0/pnpm-global/3/node_modules/.pnpm/registry.npmjs.org/microvium/0.0.8/node_modules/microvium/dist/lib/src-to-il.js:435:5)
    at compileStatement (/home/pitu/.nvm/versions/node/v13.8.0/pnpm-global/3/node_modules/.pnpm/registry.npmjs.org/microvium/0.0.8/node_modules/microvium/dist/lib/src-to-il.js:659:39)
```
Uncommenting it doesn't help as the desired effect is not achieved.

coder-mike, 2020-05-26:
This runs fine for me. Is there an entry-point I should be executing to see the error?

coder-mike, 2020-05-26:
BTW, running your example code in node.js gives me an error that `BR_v_7` is accessed before initialization. The fact that Microvium seems to permit it is technically incorrect. It's probably better to move those declarations to the top of the file. 

[image]

Another thing, when I run it in node or Microvium, it doesn't terminate.

pitust, 2020-05-27:
I think this was fixed in master.

closed 2020-05-27 by pitust

pitust, 2020-05-27:
And i moved the declaration; it runs under node now AND it runs under microvium (if i ignore the errors)