coder-mike/microvium #23: `new` support?

davidchisnall, 2022-06-21:
It appears that you can create objects via object literal syntax, but not via `new`.  Does this also mean that prototype chains don't work (it's been ages since I wrote JavaScript, but I seem to recall that `new` was the only standard way of setting the prototype of an object)?

coder-mike, 2022-06-22:
Yeah, the [code is already there](https://github.com/coder-mike/microvium/blob/main/native-vm/microvium.c#L4453) to walk the prototype chain, as you might have already noticed, which I think is the differential inheritance you're talking about. The question is how to create objects that have a prototype chain, and that's where I'm proposing a new `class` type.

I'm not sure what you mean by "less invasive". To me, this is the least invasive option.

In JS, classes are functions with a `prototype` property (and other properties corresponding to the static members of the class) and the `new` operator instantiates an object whose prototype (`__proto__`) is the prototype of the new object. But Microvium functions do not support properties and it would be very expensive to do. And I believe that in typical usage, most people don't add properties to functions that they've declared, so it's not a very useful feature outside of classes.

So that's why I'm proposing that a class is its own type -- a function-like type but which supports properties and can only be invoked using the `new` operator. This can be done by leveraging the existing mechanics of objects and functions by making a class simply a tuple of an object and a function and delegating the behavior to each accordingly. And then I propose making the `prototype` property special so that `new` is not an `O(n)` operation. Although arguably that doesn't add much benefit since then property access is an O(n) operation anyway. If you want guaranteed O(1) access then use a closure instead of an object.

davidchisnall, 2022-06-23:
I see.  I was assuming that classes would have a fixed structure with their properties mapped to fixed indexes, rather than being represented in the same way.  This is how ActionScript worked and I assumed that JavaScript classes worked the same way, but it looks as if that was wrong.  With my misunderstanding corrected, I think your approach sounds great.

coder-mike, 2022-06-23:
Great! I optimistically have opcodes implemented for reading and writing to a literal index in a fixed-length array structure, on the hope that a good optimizer will be able to convert most objects to fixed-length arrays instead. But these opcodes are currently unused.

coder-mike, 2022-09-02:
Microvium [now supports classes](https://coder-mike.com/blog/2022/09/03/microvium-has-classes/).

closed 2022-09-02 by coder-mike