coder-mike/microvium #25: npm run build fails in sync-opcodes.ts

bshanks, 2022-06-21:
$ npm --version
8.5.1
$ cd /tmp
$ git clone https://github.com/coder-mike/microvium.git
$ cd microvium
$ npm install typescript
$ ts-node --version
v9.1.1
$ npm run build

> microvium@0.0.18 build
> ts-node scripts/sync-opcodes.ts && node-gyp build --debug && tsc && ts-node scripts/preprocess-microvium && npm run copy-files && npm run size-check

/tmp/microvium/scripts/sync-opcodes.ts:74
    throw new Error('Error synchronizing. Source pattern not found');
          ^
Error: Error synchronizing. Source pattern not found
    at copyInto (/tmp/microvium/scripts/sync-opcodes.ts:74:11)
    at Object.<anonymous> (/tmp/microvium/scripts/sync-opcodes.ts:64:16)
    at Module._compile (internal/modules/cjs/loader.js:999:30)
    at Module.m._compile (/tmp/microvium/node_modules/ts-node/src/index.ts:858:23)
    at Module._extensions..js (internal/modules/cjs/loader.js:1027:10)
    at Object.require.extensions.<computed> [as .ts] (/tmp/microvium/node_modules/ts-node/src/index.ts:861:12)
    at Module.load (internal/modules/cjs/loader.js:863:32)
    at Function.Module._load (internal/modules/cjs/loader.js:708:14)
    at Function.executeUserEntryPoint [as runMain] (internal/modules/run_main.js:60:12)
    at main (/tmp/microvium/node_modules/ts-node/src/bin.ts:227:14)

coder-mike, 2022-06-21:
If I try to reproduce your steps, I get a different error:

```
gyp ERR! stack Error: You must run `node-gyp configure` first!
```

So I'm not sure quite how it got so far without node-gyp configure. But anyway, can you please try the following commands:

```
cd /tmp
git clone https://github.com/coder-mike/microvium.git
cd microvium
npm install      # <-- Not just typescript
npm run build
```

The `npm install` will install `ts-node` locally as well, so you shouldn't need it globally. It should also trigger the node-gyp configuration. Let me know if you're still running into issues.

bshanks, 2022-06-21:
Thanks, the issue still occurs:

$ npm install
npm WARN deprecated @types/immutable@3.8.7: This is a stub types definition for Facebook's Immutable (https://github.com/facebook/immutable-js). Facebook's Immutable provides its own type definitions, so you don't need @types/immutable installed!
npm WARN deprecated mkdirp@0.5.3: Legacy versions of mkdirp are no longer supported. Please update to mkdirp 1.x. (Note that the API surface has changed to use Promises in 1.x.)
npm WARN deprecated har-validator@5.1.3: this library is no longer supported
npm WARN deprecated flat@4.1.0: Fixed a prototype pollution security issue in 4.1.0, please upgrade to ^4.1.1 or ^5.0.1.
npm WARN deprecated debug@3.2.6: Debug versions >=3.2.0 <3.2.7 || >=4 <4.3.1 have a low-severity ReDos regression when used in a Node.js environment. It is recommended you upgrade to 3.2.7 or 4.3.1. (https://github.com/visionmedia/debug/issues/797)
npm WARN deprecated uuid@3.4.0: Please upgrade  to version 7 or higher.  Older versions may use Math.random() in certain circumstances, which is known to be problematic.  See https://v8.dev/blog/math-random for details.
npm WARN deprecated request@2.88.2: request has been deprecated, see https://github.com/request/request/issues/3142

> microvium@0.0.18 install
> node-gyp-build

gyp info it worked if it ends with ok
gyp info using node-gyp@6.1.0
gyp info using node@12.22.9 | linux | x64
gyp info find Python using Python version 3.10.4 found at "/usr/bin/python3"
gyp info spawn /usr/bin/python3
gyp info spawn args [
gyp info spawn args   '/tmp/microvium/node_modules/node-gyp/gyp/gyp_main.py',
gyp info spawn args   'binding.gyp',
gyp info spawn args   '-f',
gyp info spawn args   'make',
gyp info spawn args   '-I',
gyp info spawn args   '/tmp/microvium/build/config.gypi',
gyp info spawn args   '-I',
gyp info spawn args   '/tmp/microvium/node_modules/node-gyp/addon.gypi',
gyp info spawn args   '-I',
gyp info spawn args   '/home/bshanks/.cache/node-gyp/12.22.9/include/node/common.gypi',
gyp info spawn args   '-Dlibrary=shared_library',
gyp info spawn args   '-Dvisibility=default',
gyp info spawn args   '-Dnode_root_dir=/home/bshanks/.cache/node-gyp/12.22.9',
gyp info spawn args   '-Dnode_gyp_dir=/tmp/microvium/node_modules/node-gyp',
gyp info spawn args   '-Dnode_lib_file=/home/bshanks/.cache/node-gyp/12.22.9/<(target_arch)/node.lib',
gyp info spawn args   '-Dmodule_root_dir=/tmp/microvium',
gyp info spawn args   '-Dnode_engine=v8',
gyp info spawn args   '--depth=.',
gyp info spawn args   '--no-parallel',
gyp info spawn args   '--generator-output',
gyp info spawn args   'build',
gyp info spawn args   '-Goutput_dir=.'
gyp info spawn args ]
gyp info spawn make
gyp info spawn args [ 'BUILDTYPE=Release', '-C', 'build' ]
make: Entering directory '/tmp/microvium/build'
  CC(target) Release/obj.target/nothing/node_modules/node-addon-api/src/nothing.o
  AR(target) Release/obj.target/node_modules/node-addon-api/src/nothing.a
  COPY Release/nothing.a
  CXX(target) Release/obj.target/native-vm/native-vm-bindings/index.o
In file included from ../native-vm-bindings/../native-vm/microvium_port_test.h:11,
                 from ../native-vm-bindings/microvium_port.h:1,
                 from ../native-vm-bindings/../native-vm/microvium.h:14,
                 from ../native-vm-bindings/Value.hh:4,
                 from ../native-vm-bindings/index.cc:1:
../native-vm-bindings/../native-vm/microvium_port_example.h:235:17: warning: ‘uint16_t crc16(void*, uint16_t)’ defined but not used [-Wunused-function]
  235 | static uint16_t crc16(MVM_LONG_PTR_TYPE lp, uint16_t size) {
      |                 ^~~~~
  CXX(target) Release/obj.target/native-vm/native-vm-bindings/NativeVM.o
../native-vm-bindings/NativeVM.cc: In member function ‘void VM::NativeVM::runGC(const Napi::CallbackInfo&)’:
../native-vm-bindings/NativeVM.cc:127:8: warning: variable ‘env’ set but not used [-Wunused-but-set-variable]
  127 |   auto env = info.Env();
      |        ^~~
In file included from ../native-vm-bindings/../native-vm/microvium_port_test.h:11,
                 from ../native-vm-bindings/microvium_port.h:1,
                 from ../native-vm-bindings/../native-vm/microvium.h:14,
                 from ../native-vm-bindings/NativeVM.hh:6,
                 from ../native-vm-bindings/NativeVM.cc:3:
../native-vm-bindings/../native-vm/microvium_port_example.h: At global scope:
../native-vm-bindings/../native-vm/microvium_port_example.h:235:17: warning: ‘uint16_t crc16(void*, uint16_t)’ defined but not used [-Wunused-function]
[... 197 lines cut ...]
added 338 packages, and audited 339 packages in 7s

42 packages are looking for funding
  run `npm fund` for details

found 0 vulnerabilities
$ npm run build

> microvium@0.0.18 build
> ts-node scripts/sync-opcodes.ts && node-gyp build --debug && tsc && ts-node scripts/preprocess-microvium && npm run copy-files && npm run size-check

/tmp/microvium/scripts/sync-opcodes.ts:74
    throw new Error('Error synchronizing. Source pattern not found');
          ^
Error: Error synchronizing. Source pattern not found
    at copyInto (/tmp/microvium/scripts/sync-opcodes.ts:74:11)
    at Object.<anonymous> (/tmp/microvium/scripts/sync-opcodes.ts:64:16)
    at Module._compile (internal/modules/cjs/loader.js:999:30)
    at Module.m._compile (/tmp/microvium/node_modules/ts-node/src/index.ts:858:23)
    at Module._extensions..js (internal/modules/cjs/loader.js:1027:10)
    at Object.require.extensions.<computed> [as .ts] (/tmp/microvium/node_modules/ts-node/src/index.ts:861:12)
    at Module.load (internal/modules/cjs/loader.js:863:32)
    at Function.Module._load (internal/modules/cjs/loader.js:708:14)
    at Function.executeUserEntryPoint [as runMain] (internal/modules/run_main.js:60:12)
    at main (/tmp/microvium/node_modules/ts-node/src/bin.ts:227:14)

bshanks, 2022-06-21:
I'm on GNU/Linux pop-os 22.04. The offending call to copyInto in sync-opcodes.ts is:

tsSnapshotIL = copyInto(cMicroviumBytecode, tsSnapshotIL,
  '#define MVM_BYTECODE_VERSION ', '\r\n',
  'export const BYTECODE_VERSION = ', ';');

I think the \r\n is the Windows-specific newline delimiter, perhaps that's messing it up?

coder-mike, 2022-06-21:
Yes, that last error is benign. I decided that because an objective Microvium is to be small, I would measure the size after each build.

coder-mike, 2022-06-21:
Thanks for your help diagnosing this. I believe I've fixed both issues now -- just a `chmod` for those scripts, and the change to the newline to support both Windows and Linux. Would you mind running everything again and seeing if it works now?

bshanks, 2022-06-21:
It gets thru most of the build, but at the end it doesn't print "Microvium flash size:  8.22 kB"
anymore, instead there are a few new errors:

> bash ./size-test.sh

/tmp/microvium/size-test /tmp/microvium
./build.sh: line 4: arm-none-eabi-gcc: command not found
./build.sh: line 14: arm-none-eabi-objdump: command not found
./build.sh: line 15: arm-none-eabi-objdump: command not found
./build.sh: line 16: arm-none-eabi-size: command not found

bshanks, 2022-06-21:
I'd also suggest mentioning "npm install" in doc/contribute.md

coder-mike, 2022-06-22:
Ok, the new errors actually show it's getting further :-)

What's happening is that the size measurement is using the GCC compiler for the Cortex M0. Maybe I should make that step conditional on an environment variable.

coder-mike, 2022-06-24:
I've made a note `contribute.md`. I also recommend installing the [ARM GNU Toolchain](https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/downloads) so that the size tests work, if your intention is to contribute, since we want to keep a close eye on how the changes affect the size.

closed 2022-06-24 by coder-mike