coder-mike/microvium #28: mvm_typeOf returns surprising values

davidchisnall, 2022-06-22:
If I create an object as an object literal (`{}`) or an array as an array literal (`[]`) in JavaScript then I can add properties to the object and elements to the array as I expect.  If I then pass these as an argument to C code, then I can call `mvm_typeOf` to query the type, but I always see `VM_T_UNDEFINED` as the return value.

coder-mike, 2022-06-22:
Indeed, that is surprising and sounds like a bug. Do you have a snippet that reproduces this?

coder-mike, 2022-06-22:
Oh 🤦‍♂️ It's not a bug, it's just a surprising behavior. See the [limitations in the readme](https://github.com/coder-mike/microvium/blob/main/readme.md#limitations)... only simple types are supported across the FFI. I'll elaborate in another comment.

coder-mike, 2022-06-22:
I agree that this is confusing, and so probably the behavior needs to change, but I'm not entirely sure what the behavior should look like. There are a few options.

To elaborate on why this decision was made to have the behavior like this...

I have an experimental optimizer in the works called Microvium Boost which does whole-program static analysis on a snapshot ([see here](https://coder-mike.com/blog/2020/06/27/microvium-boost/)). It can do things like infer what properties on an object are not used. This kind of analysis is a bit simpler and produces better results if references can't cross the FFI boundary. Basically, any object that crosses the FFI boundary is unoptimizable, and this is contagious to other objects as well (e.g. if `obj1` is not optimizable because it's passed across the boundary for C to have its way with it, then `obj2[obj1.p] = 42` makes `obj2` unoptimizable as well). Of course, this is not just a characteristic of Microvium Boost but a general issue that any future optimizer for Microvium will face, so the design decision was to avoid painting Microvium into a corner in future. But maybe it was the wrong decision, I'm not sure.

I strongly value ease of use, so this needs a solution. 

## Solution 1: no change

People can deal with this with workarounds. As long as there is a way for information to cross the boundary, e.g. in the form of simple primitive values, then there is a way to marshal anything you like across the boundary, with enough effort.

## Solution 2: just don't do the sanitization

Allow references to JS objects to be passed across the FFI boundary, and add API support for reading and writing these objects (or creating new ones).

## Solution 3: allow the passing C structs rather than JS objects across the boundary

I could add first-class support for JS objects to be serialized to C structs on the JS heap and passed across the boundary, and to allow C to create these structures on the JS heap. Or to have support for byte arrays in JS which allows JS code to serialize an object to a byte array (or a library) and then pass that across the boundary.

## Solution 4: allow the passing of C pointers across the boundary

Or flipping it the other direction, I could add a Microvium type that wraps a C pointer, and have builtin Microvium operations for manipulating C pointers (reading and writing). These could be paired with a size and reference count that allows the engine to enforce reading and writing within certain bounds and prevents unintentional or malicious access through dangling pointers. The difference between #3 and #4 is that #4 has the structs as mallocs in C-land while #3 has the structs on the GC heap.

----------------

#4 requires sanitization during a snapshot from C since the pointers should probably not survive the snapshot/restore cycle. #3 does not require this sanitization. 

#2 and #3 require the user to understand handles and the GC, whereas #4 does not. That on its own is a huge argument in favor of #4. 

I also have in mind that Microvium Boost eventually does binary type inference (it's already keeping track of most of the information required to do this) which would open the door to Microvium using LLVM or something to actually compile sequences of JS to native code automatically, which would be a step towards MetalScript. This would be easier with solution #4 since it would give the JS code the power to intervene with C data rather than giving C the power to intervene with JS data. (I don't have much documentation on MetalScript, but if you're interested there's a PowerPoint slide about halfway down [this post](https://coder-mike.com/blog/2018/10/23/tc53-meeting/)). 

--------

So anyway, that's my thought process so far. I know that this long comment is probably only relevant to a small audience. If anyone reading this has any thoughts or suggestions, let me know. Also, give me a sense of priority if this is something you need for your use case, since I've got a long list other things I'm also adding to Microvium :-) .

garthk, 2022-06-23:
The last time I did this kind of thing with V8 I was wading through XML, filtering, and creating a JS summary object from C++ side because it was the call count that was hurting. For uVM though it strikes me you're better off with the closer-to-the-hardware code managing the struct's allocation and the uVM-side handle object making it easy to pull out simple values at fixed offsets. Maybe provide a way for the C to get called back if the handle gets collected?

davidchisnall, 2022-06-23:
Thanks.  I hit this because I was trying to make the print function in the example behave more like the browser one and log `[object]` for objects and so on.  I was quite surprised that it showed objects as `undefined` in the existing to-UTF8 function and then discovered that it did the same for later objects.

I'd like to be able to provide a C interface to some structured RPC mechanism, which would require being able to parse a structure from some wire format, construct an object with that representation, and then hand them to JavaScript, and then in reverse to receive an object, construct a wire-format representation, and send it.  This precludes anything that requires me to specify the structure that I'm passing in, because it's data dependent.

There are a couple of things that make this easier for a static analyser.  The output function takes, effectively, immutable object pointers.  If, when I register the function on the JavaScript side, I make this promise, then the static analysis is free to assume it.  In a debug build, you could set a flag during FFI to say that arguments are immutable and assert if any of them are modified.  

My goal here is to be able to upgrade the data formats and the JavaScript together, without modifying the C code.

The object passed in from the outside is, as a result of the requirements, of an unknown structure.  There isn't anything that an optimiser can do with that, but I'm happy with opting out of optimisations there.

This is more or less your option 2, but with the extra opt-in immutability guarantee.  

For some operations, I am happy to mandate the structure that's passed back, for example as a JavaScript class once that's available, and let the optimiser assume that and do the wrong thing if I don't give a class with the right set of fields provided.

coder-mike, 2022-06-25:
@davidchisnall would it work at all for you if the wire format were interpreted/serialized on the JS side rather than the C side? So then only the buffer needs to pass through the FFI boundary (which is also not yet supported)

The reason I'm exploring that side is mainly just the pitfalls of managing GC handles in C-land. Like, if someone calls a hypothetical mvm_createObject function and then mvm_setProperty, they may not realize that mvm_setProperty has the potential to trigger a GC collection and thus free up the object they just created. And worse is that this kind of bug could sit dormant for ages because most calls to mvm_setProperty won't trigger the collection. And it's not a fail-fast kind of bug either. Since the effect could just cause random corruption and strange things happening.

coder-mike, 2022-06-25:
Alternatively I can imagine that in safe-mode it could validate that every reference passed in through the FFI is also registered as a GC-reachable handle.  I wonder if that's a better way to go

coder-mike, 2022-06-27:
WIP: on branch `28-support-objects-in-microvium` I've removed the argument sanitization so now `mvm_typeOf` should return the exected type codes. Passing test cases [here](https://github.com/coder-mike/microvium/blob/0809fc85f62a49047c414388d5055a9229421bda/test/native/native-api.test.ts#L8-L61).

coder-mike, 2022-06-28:
I was thinking of adding new API methods to the Microvium C API for creating objects and reading/writing properties, but then I realized that this isn't necessary since you can write your own JS functions to do this and export them for your C code to access.

```js
const newObject = () => ({});
const getProp = (o, p) => o[p];
const setProp = (o, p, v) => o[p] = v;

vmExport(1, newObject);
vmExport(2, getProp);
vmExport(3, setProp);
```

See [test case / example here](https://github.com/coder-mike/microvium/blob/46b9037cfdc239925ab088f74ac7223e7c251bb6/test/native/native-api.test.ts#L63-L69)

coder-mike, 2022-06-28:
I've merged this all into `main`. Going to mark this ticket as done but let me know if you have any issues.

closed 2022-06-28 by coder-mike

davidchisnall, 2022-06-29:
Probably needs a new issue, but the big limitation is that I can't pass arrays (especially arrays of primitives) across the boundary.  If I need to do an invocation for each array element then that's going to be very slow.  If I can serialise something in JavaScript then pass it to C, that's very flexible.

davidchisnall, 2022-06-29:
Oh, the other thing here is that I can't *enumerate* properties from C.  I need to know every property that the object exposes, which means I can't do something like a to-JSON function, for example.