coder-mike/microvium #30: VS won't compile

madmax911, 2022-06-24:
Hi,

Just wanted to give your program a look but when I went to open the native-vm-vs-project Visual Studio solution it had many errors about missing .h files and many others.  I'll attach a screenshot.

[image]

coder-mike, 2022-06-24:
Did you have a look at the getting-started guide ([doc/getting-started.md](https://github.com/coder-mike/microvium/blob/main/doc/getting-started.md))? The project you're trying run here is not an example of how to use Microvium, it's project used to develop and test Microvium.

Having said that, I'm able to reproduce what you're seeing if I use a clean clone, so something is not quite right. I'll look into it.

coder-mike, 2022-06-24:
Ok, I've made some changes to fix the compile errors. I think VS was somehow storing some additional state in the `.vs` folder which is not checked into the repo, since if I deleted that folder then it loses the project settings (in this case the include paths and C++ version)

But I still recommend going to [doc/getting-started.md](https://github.com/coder-mike/microvium/blob/main/doc/getting-started.md) if you're just trying it out.

Maybe at some point I need an "examples" folder with pre-baked project files for a couple of different cases.

closed 2022-06-24 by coder-mike

madmax911, 2022-06-27:
Thanks Mike,

I looked at the readme a little - honestly, I'm new to it and just checking it out for curiosity sake.