coder-mike/microvium #32: Enumerate properties from C

coder-mike, 2022-06-29:
Copied from [#28](https://github.com/coder-mike/microvium/issues/28#issuecomment-1169729380)

I can't enumerate properties from C. I need to know every property that the object exposes, which means I can't do something like a to-JSON function, for example.

@davidchisnall

coder-mike, 2022-06-29:
I agree that this would be very useful, and this kind of reflection is not possible without built-in support for it. This is roughly my idea so far:

My plan here is to implement [Reflect.ownKeys](https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Global_Objects/Reflect/ownKeys), which returns an array of all the keys of an object. 

I intend to support it only for Microvium objects and not including arrays -- it will produce a type error if done on an array. This still fits the goal of Microvium supporting a *subset* of JS.

My reasoning for supporting `Reflect.ownKeys` rather than `Object.keys` or `Object.getOwnPropertyNames` or `for ... in`. is that `Reflect.ownKeys` returns all the object keys regardless of their type (symbol or string) or attributes (enumerable or not enumerable). In Microvium at the moment, keys are only strings, but if it supports symbols in the future then I don't need to filter these from the `Reflect.ownKeys` results. Microvium properties do not have attributes -- all properties are treated as `writable`, and the enumerability of properties is not observable. Using `Reflect.ownKeys` means that I preserve the fact that enumerability is not observable, which allows me to add both enumerable and non-enumerable properties without actually keeping track of which are which.

My reasoning to do this in JS-land instead of the C API is:

  1. I want to keep the C API lean. This is because an objective of the Microvium design is to be easy to integrate. A leaner API is less daunting and easier to get your head around.
  2. I don't want to give more power to C than to JS. If C can reflect on JS object keys, then JS should be able to reflect on its own keys as well.
  3. `Reflect.ownKeys` is part of the ECMAScript standard, whereas there is no standard for how to do this in a C API.
  4. Although JS can call C and vice versa, so a function on one side could be made accessible to the other side, having it in JS-land means the optimizer can directly understand it and optimize it.

But I'm open to arguments on the other side as well.

coder-mike, 2022-06-29:
To give more detail on the plan, for anyone interested (but more just to document this for myself):

  - I'm going to introduce a new opcode for this. The pattern in Microvium so far is that builtin engine behavior is accessible through opcodes (there are no builtin functions or libraries in the engine that are accessible to JS). 
  - This also means that optimizers only need to consider the instruction set, and not the behavior of a separate builtin library. 
  - Microvium has existing opcodes for objects: `ObjectGet`, `ObjectSet` and `ObjectNew`. This new opcode may be named `ObjectKeys`, to follow the pattern.
  - To make this opcode accessible to JS code, we need a new global `Reflect` and a new function `ownKeys` which wraps the new opcode. I plan on creating these during the initialization of the compile-time virtual machine.

davidchisnall, 2022-06-30:
The combination of this and the ability to pass byte buffers to/from C would make it possible for me to write arbitrary [de]serialisation code in JavaScript.

coder-mike, 2022-06-30:
Microvium now has a `Reflect.ownKeys` global builtin. 

```js
const obj = { a: 1, b: 2 };
Reflect.ownKeys(obj); // returns ['a', 'b']
```

If you need to access it from C, you can export it:

```js
vmExport(1, Reflect.ownKeys)
```

closed 2022-06-30 by coder-mike