coder-mike/microvium #39: FEATURE: minimal parser and bytecode generation, for browser bundling

morganrallen, 2022-09-21:
Howdy Mike, I'm revisiting some of my ESP32 work and have seen a future roadblock for one of my goals.

One of the things I have in mind is the ESP32 serving up an editor over WiFi. From there a user can manage and edit scripts, to be either updated on the fly or deployed once ready. In either case it requires being able to parse the JS source. Since the C interface has no parser, this cannot be done on the backed. 
Additionally it seems the microvium module includes too many NodeJS specific modules to be bundled. I've tried requiring more specific portions of microvium, like `virtual-machine` but seems this doesn't do a lot of setup around `globalSlots`, `globalVariables`, etc. This is handled by `virtual-machine-friendly`, which unfortunately imports the core `Microvium` module.... repeat module import errors...

I've started down the rabbit hole of trying to bootstrap those values, but this seems the wrong approach, especially with my limited TS experience. Though I do believe this can be broken down in a way that doesn't require OS level modules, possibly shifting these imports to the CLI interface while allowing a more specific `require` call when needed. I may even have missed something, so any input on this would be appreciated.

coder-mike, 2022-09-21:
Hi @morganrallen. 

> Additionally it seems the microvium module includes too many NodeJS specific modules to be bundled

I'm not sure I understand. If your intention is to host a web-based editor on an ESP32, then what purpose does bundling the compiler do? If the result is a bundle of JS code, then you're back to square one requiring a JS runtime that can parse and run that code (which the Microvium runtime would not be able to do since it only supports a subset of the language). If your plan was to bootstrap the Microvium front-end to run on the Microvium backend, I don't expect this to work since Microvium only supports a subset of the JS spec. It would be awesome to someday have it self-hosting, but that's a long way off since the front-end compiler is a beast.

> In either case it requires being able to parse the JS source.

If you need a parser at runtime on an embedded device, and you're using a beefy device like ESP32, I might suggest looking into the [Moddable XS engine](https://github.com/Moddable-OpenSource/moddable). It has a built-in parser at runtime and there are examples in the SDK on how to use it on ESP32. Also, this is the smallest engine that supports the full modern ECMAScript spec, which will make it easier to set up a web server. And it has its own HTTP module with [examples on how to use it](https://github.com/Moddable-OpenSource/moddable/blob/public/examples/network/http/httpserver/main.js).

If you're looking for something smaller than XS but which has a parser at runtime, mJS and Elk are options but they're quite limited in terms of supported language features.

If I'm misunderstanding your use case, let me know.

coder-mike, 2022-09-21:
Oh, is your intention to run the compiler in the browser client and have the client post the snapshot to the ESP32 server to execute?

morganrallen, 2022-09-21:
> Oh, is your intention to run the compiler in the browser client and have the client post the snapshot to the ESP32 server to execute?

Yes, sorry that wasn't clear, I started writing this very late last night, should have just started over fresh this morning.

coder-mike, 2022-09-21:
Ok, that makes sense. That would be cool to see! 

Two thoughts come to mind:

  - Can you shim out the node.js API (e.g. with something equivalent to [rollup-plugin-node-builtins](https://github.com/calvinmetcalf/rollup-plugin-node-builtins)).
  - If it's getting stuck with `VirtualMachineFriendly` then possibly it's the `SynchronousWebSocketServer` which uses a few Node modules including `worker_threads`. This is unused at the moment so we could just comment out that code for the moment and I can revisit it later (it's WIP for adding Microvium debugger), if it doesn't stub well.

morganrallen, 2022-09-21:
Yes, it hangs up the the WS server for sure. My preferred bundler is `browserify` and it outputs...

```
$ echo "mcvm = require('microvium');"|./node_modules/.bin/browserify -
Error: Can't walk dependency graph: ENOENT: no such file or directory, lstat './worker_threads'
    required by /home/morgan/devel/lorajs/node_modules/microvium/dist/lib/synchronous-ws-server/index.js
```

coder-mike, 2022-09-21:
I've [commented out that dependency](https://github.com/coder-mike/microvium/commit/b7139526420c97f19aca65bcc429c892d5604f00). Does that help?

morganrallen, 2022-09-21:
Yup, that'll get it to build.
```
$ echo "mcvm = require('microvium');"|./node_modules/.bin/browserify -|wc
88377  300926 2845183
```

coder-mike, 2022-09-22:
Ok, great. Glad it worked for you!

closed 2022-09-22 by coder-mike