coder-mike/microvium #40: Document *result / C return values (generally more C embedding docs)

morganrallen, 2022-10-11:
Hia @coder-mike, doing a bit of hacking again and I cannot seem to find documentation on returning values from C function call (JS calling C). I've tried a couple different way but typically hit assertion errors. Can you provide a snippet showing how this should work?

coder-mike, 2022-10-11:
Yes, good point. I'll add that. In the meantime, here's an example that should get you going:

```js
const print = vmImport(0);
const add = vmImport(1);
vmExport(0, entry);

function entry() {
  const result = add(1, 2);
  print(`The result is ${result}`);
}
```

```c
#include <stdio.h>
#include "microvium.h"

const uint8_t snapshot[] = {0x04,0x1c,0x03,0x00,0x8c,0x00,0xdc,0xf7,0x03,0x00,0x00,0x00,0x1c,0x00,0x20,0x00,0x24,0x00,0x24,0x00,0x28,0x00,0x2a,0x00,0x7c,0x00,0x82,0x00,0x00,0x00,0x01,0x00,0x00,0x00,0x61,0x00,0x09,0x00,0x81,0x00,0x2d,0x00,0x05,0x40,0x70,0x75,0x73,0x68,0x00,0x00,0x0f,0x40,0x54,0x68,0x65,0x20,0x72,0x65,0x73,0x75,0x6c,0x74,0x20,0x69,0x73,0x20,0x00,0x00,0x00,0x00,0x02,0x60,0x00,0x00,0x02,0x60,0x01,0x00,0x0d,0x50,0x04,0x31,0x30,0x30,0x88,0x21,0x00,0x6b,0x12,0x6f,0x67,0x01,0x60,0x00,0x1c,0x50,0x05,0x88,0x1d,0x00,0x89,0x01,0x00,0x01,0x07,0x08,0x78,0x03,0xa0,0x89,0x00,0x00,0x01,0x88,0x35,0x00,0x13,0x6c,0x78,0x02,0x67,0x67,0x01,0x60,0x49,0x00,0x4d,0x00,0x02,0x00,0x08,0xc0,0x09,0x00,0x09,0x00,0x2d,0x00,0x51,0x00};

mvm_TeError resolveImport(mvm_HostFunctionID hostFunctionID, void* context, mvm_TfHostFunction* out_hostFunction);
mvm_TeError print(mvm_VM* vm, mvm_HostFunctionID hostFunctionID, mvm_Value* result, mvm_Value* args, uint8_t argCount);
mvm_TeError add(mvm_VM* vm, mvm_HostFunctionID hostFunctionID, mvm_Value* result, mvm_Value* args, uint8_t argCount);

int main() {
  mvm_VM* vm;
  mvm_TeError err;
  mvm_VMExportID exportIDs[] = { 0 };
  mvm_Value exports[sizeof exportIDs / sizeof exportIDs[0]];

  err = mvm_restore(&vm, (void*)snapshot, sizeof snapshot, NULL, resolveImport);
  if (err != MVM_E_SUCCESS) return err;
  err = mvm_resolveExports(vm, exportIDs, exports, sizeof exports / sizeof exports[0]);
  if (err != MVM_E_SUCCESS) return err;

  mvm_Value entry = exports[0];
  err = mvm_call(vm, entry, NULL, NULL, 0);
  if (err != MVM_E_SUCCESS) return err;
}

mvm_TeError resolveImport(mvm_HostFunctionID hostFunctionID, void* context, mvm_TfHostFunction* out_hostFunction) {
  switch (hostFunctionID) {
    case 0: *out_hostFunction = print; return MVM_E_SUCCESS;
    case 1: *out_hostFunction = add; return MVM_E_SUCCESS;
  }
  return MVM_E_FUNCTION_NOT_FOUND;
}

mvm_TeError print(mvm_VM* vm, mvm_HostFunctionID hostFunctionID, mvm_Value* result, mvm_Value* args, uint8_t argCount) {
  if (argCount < 1) return MVM_E_HOST_ERROR;
  printf("%s\n", mvm_toStringUtf8(vm, args[0], NULL));
  return MVM_E_SUCCESS;
}

mvm_TeError add(mvm_VM* vm, mvm_HostFunctionID hostFunctionID, mvm_Value* result, mvm_Value* args, uint8_t argCount) {
  if (argCount < 2) return MVM_E_HOST_ERROR;
  int a = mvm_toInt32(vm, args[0]);
  int b = mvm_toInt32(vm, args[1]);
  int r = a + b;
  *result = mvm_newInt32(vm, r);
  return MVM_E_SUCCESS;
}
```

Let me know if you run into any other issues!

TobyEalden, 2022-10-12:
Is it possible to return an object?

I've just started looking into microvium - it's pretty impressive, nice work!

morganrallen, 2022-10-12:
ok I was on the right track, I've got a bug somewhere else I cannot figure out.
And for some clarity, if I'm seeing this right. `mvm_Value` is just a 16-bit int, does this end up representing a the memory location of that value inside the VM?

coder-mike, 2022-10-12:
> Is it possible to return an object?

@TobyEalden Yep. I think you found the [the unit test/example](https://github.com/coder-mike/microvium/blob/46b9037cfdc239925ab088f74ac7223e7c251bb6/test/native/native-api.test.ts#L61-L105). I'll have a look at the specific code you posted in the other ticket.

> I've just started looking into microvium - it's pretty impressive, nice work!

Thank you!

> I've got a bug somewhere else I cannot figure out.

@morganrallen If you want to share your code, I'm happy to look at it and see if I can help.

> And for some clarity, if I'm seeing this right. mvm_Value is just a 16-bit int, does this end up representing a the memory location of that value inside the VM?

In short, yes. You can treat it as a reference to Microvium's garbage-collected heap. Internally, it's a [bit more complicated](https://github.com/coder-mike/microvium/blob/main/native-vm/microvium_internals.h#L22-L167) because, for efficiency reasons, certain kinds of values are actually inlined into the 16-bit int rather than being allocated on the heap, and because it can either reference the garbage-collected heap or the bytecode image that you passed to `mvm_restore`. But for practical purposes, you can ignore all that and treat it as if it's a reference to the VM's memory every time.

Be aware that Microvium uses a compacting garbage collector, so memory may move during a GC collection cycle and so the `mvm_Value` values need to be updated to point to the new locations. See [Memory Management](https://github.com/coder-mike/microvium/blob/main/doc/native-host/memory-management.md).

As you ask questions, I'll take note to add any clarifications to the docs as well so it's easier for other people.

morganrallen, 2022-10-13:
Going to close this as it turns out there was a silly bug on my part, but increasing documentation around this would still be helpful.

closed 2022-10-13 by morganrallen