coder-mike/microvium #41: Support for populating objects in C/C++

TobyEalden, 2022-10-12:
I'm trying to create an object in C land, populate it and return it to the VM. Minimal reproduction is [here](https://github.com/TobyEalden/microvium-get-object).

In brief, I'm exporting `createObject` and `setProp` functions to the host, and I want to call them from the host and return a populated object to the VM for further processing. Full details at [microvium-get-object](https://github.com/TobyEalden/microvium-get-object).

This might be similar to https://github.com/coder-mike/microvium/issues/40, and maybe https://github.com/coder-mike/microvium/issues/28, but I think I'm missing something.

morganrallen, 2022-10-12:
Nice work around, though would be nice to having something like `mvm_newObject`

coder-mike, 2022-10-12:
Ok, I've found 2 bugs on your side, and it looks like there's a bug on my side as well that I need to figure out.

1. When you do `mvm_newString`, you should pass the length of the string excluding the null terminator, otherwise the null terminator becomes part of the string. E.g. it becomes the JavaScript string `"name\0"` instead of `"name"`. I'll add this to the documentation.
2. The `args` to `mvm_call` should be a pointer to an array of `mvm_Value`. In your code, `setPropArgs` is instead an array of pointers to `mvm_Value`. I've fixed it below here:
3. On my side, it looks like Microvium has a bug with [string interning](https://en.wikipedia.org/wiki/String_interning), so the literal string "name" in Microvium has different semantics as a property key than the string `mvm_newString(vm, "name", 4)`. So it's a bit weird, but if you change the `console.log` line in `main` to `console.log(res['n' + 'ame']);` then it works, but if you use `res.name` then it doesn't work. I'm looking into this issue now. 

```cpp   
    mvm_Value propName = mvm_newString(vm, "name", 4);
    mvm_Value propValue = mvm_newString(vm, "Toby", 4);

    mvm_Value setPropArgs[3] = {*resourceObj, propName, propValue};

    mvm_Value propResult;
    if ((err = mvm_call(vm, setProp, &propResult, setPropArgs, 3)) !=
        MVM_E_SUCCESS) {
        std::cout << "mvm failure calling setProp: " << err << std::endl;
        return err;
    }
```

coder-mike, 2022-10-13:
> Nice work around, though would be nice to having something like mvm_newObject

I intend to develop an FFI library for stuff like this. That may take a few months to get out since I'm also working on async-await which is a beast. But in the meantime, you could add it yourself with something like this library I've just hacked together now:

[https://github.com/coder-mike/microvium-ffi-example](https://github.com/coder-mike/microvium-ffi-example)

I haven't tested it yet -- I'll test and debug it when I get more time.

The use of the example in [mvm_newObject.js](https://github.com/coder-mike/microvium-ffi-example/blob/main/mvm_newObject.js) shows how you can augment Microvium to have these missing methods, without having to change the Microvium core implementation or API. This gives the potential to keep the core lean, and a user can bring in libraries like this when they need specific features like `mvm_newObject`.

@morganrallen this also relates to your [other ticket](https://github.com/coder-mike/microvium/issues/40) about documenting the FFI. My ultimate plan is to abstract it away with a proper FFI library like this to make it easier to use.

TobyEalden, 2022-10-13:
@coder-mike thanks for putting this together - much appreciated.

I'm just beginning to get my head around the snapshot paradigm, it requires a bit of a mind shift.

On initial inspection, it looks like the `mvm_newObject` export should return `any` [here](https://github.com/coder-mike/microvium-ffi-example/blob/main/mvm_newObject.js#L3), but I'll have a proper play with it today.

coder-mike, 2022-10-13:
> On my side, it looks like Microvium has a bug with [string interning](https://en.wikipedia.org/wiki/String_interning)

I've now fixed this. Version 0.0.23 on npm has this fix (latest `main`). It was a bug in the compiler, that it wasn't generating the whole string intern table so some strings were interning incorrectly.

> On initial inspection, it looks like the mvm_newObject export should return any [here](https://github.com/coder-mike/microvium-ffi-example/blob/main/mvm_newObject.js#L3)

Yes, good point.

closed 2022-10-21 by TobyEalden