coder-mike/microvium #42: asserstion failure in ShortPtr_decode when passing value return from VM into VM function call

morganrallen, 2022-10-13:
very minimal example, `print` and `fgets` are both defined in C. Calling `print("somestring");` works as expected but when passing a value returned from `fgets` it fails. Full test here. https://git.oit.cloud/morgan/mvm-tests

```
const print = vmImport(0);
const fgets = vmImport(1);
print(fgets());
```

morganrallen, 2022-10-13:
Of Note: This behaves slightly differently from the ESP-IDF code I've been working on, in that it doesn't trigger the issue until return is press, on ESP32 it happens on any key press (as expected)

coder-mike, 2022-10-13:
Another note. Handles can be really confusing. I find the most confusing thing about the node native API is [handles and handle scopes](https://nodejs.org/api/n-api.html#object-lifetime-management). I've tried to explain it as best I can in that [memory management document](https://github.com/coder-mike/microvium/blob/main/doc/native-host/memory-management.md) but if you have any suggestions for improvements to the docs, let me know.

coder-mike, 2022-10-13:
Actually, I could probably make the `result` slot reachable by the GC. That would eliminate this exact error.

morganrallen, 2022-10-13:
*facepalm* I missed that GC call was still in there and that totally makes sense. That and the '\0' checks are remnants of debugging my prior issue.  I'm going to reread that document later.

This is a bunch of good progress though, I now have a super basic prompt using just those two bindings and a generally better understanding of the workflow.

closed 2022-10-13 by morganrallen