coder-mike/microvium #43: VM doesn't make a copy of the snapshot

TobyEalden, 2022-10-21:
Not sure if this is by design, but it caught me out. It seems that `mvm_restore` doesn't make a copy of the snapshot.

For example if a reference to `vm` is kept but the `snapshot` array below goes out of scope subsequent use of `vm` will crash.

```c
err = mvm_restore(&vm, snapshot, snapshotSize, NULL, resolveImport);
```

I'm loading the snapshot at runtime from file, allocating memory for it, restoring the VM into a member variable, then deleting the snapshot. Of course I could keep the snapshot hanging around too, just seems a bit redundant.

coder-mike, 2022-10-21:
Thanks, I should have noted that in the documentation of `mvm_restore`. I'll add that now.

The reason is that the snapshot is immutable, so it can be stored in program/constant/flash memory. If it made a copy at startup, it would always be using RAM for the snapshot, which is expensive (especially on microcontrollers). Also, multiple instances of the VM can share the same snapshot in memory (you can restore multiple VMs from the same snapshot, and they can all point to the same snapshot).

Microvium is optimized for microcontrollers, which don't have a file system by default, and don't have much RAM to spare. I imagine a typical solution on an MCU will be to embed the snapshot into flash and then the VM can reference that. For example, on an MCU with 4kB of RAM and 32kB of flash, you could host maybe a 16kB snapshot image but definitely can't copy a 16kB snapshot into RAM. Even if you have a device with lots of RAM, there may be more useful things to spend it on. 

And even on a desktop-class machine, program memory can be cheaper than normal read/write memory because it can be shared between multiple processes and probably doesn't need to use swap space. Or similarly if you create a read-only memory map of the file instead of reading it into memory. Not that that matters for most cases on a desktop-class machine, but I'm thinking of a case like a cloud-based server that is able to host a million Microvium VMs that share a common pool of snapshot images -- something I might do at some point.

TobyEalden, 2022-10-22:
Thanks for the explanation @coder-mike, I figured it was something along those lines. The use case I'm exploring isn't as restricted as microcontrollers, but it's useful to know the thinking behind it.

closed 2022-10-22 by TobyEalden

coder-mike, 2022-10-22:
Out of interest, what's your use case?

TobyEalden, 2022-10-23:
I'm working on a platform that lets you securely share services and data using p2p, it's designed to run on medium to high spec routers/gateways, NAS, STB, RPi and that kind of thing (as well as desktop). I'm looking into ways we could add basic scripting capabilities.

coder-mike, 2022-10-23:
Awesome!