coder-mike/microvium #44: [feature request] add net and event modules

radiocom, 2022-10-21:
I don't know how many native functions I need to add in order to connect to some TCP ports.
we need some demos like the ffi module, thanks.

coder-mike, 2022-10-21:
> we need some demos like the ffi module, thanks.

Are you referring to [this FFI module](https://github.com/coder-mike/microvium-ffi-example)?

> I don't know how many native functions I need to add in order to connect to some TCP ports.

You would likely want to find a C or C++ library that handles networking. If you don't already have one in your firmware, I can recommend trying [Mongoose](https://mongoose.ws/documentation/). You can then integrate with the Mongoose API using something like the [Microvium FFI example library](https://github.com/coder-mike/microvium-ffi-example).

Depending on your level of experience, you could wrap Mongoose into a Microvium library for others to use as well. Similar to how the [Microvium FFI example library](https://github.com/coder-mike/microvium-ffi-example) is able to produce C++ code at compile time and then link to it at runtime, a Mongoose wrapper library can "generate" the Mongoose C++ code (i.e. just `fs.writeFileSync` to dump the Mongoose files to disk) as well as all the JavaScript-to-C++ bindings. I'll probably do this at some point if nobody else does, but since it doesn't require any additional engine capabilities, it's a library that in theory anyone can write (and I'm happy to advise and help with the design).

Having said that, even if you just do basic integration for your own project, I'm happy to help. Just let me know where you get stuck.

davidchisnall, 2022-11-24:
@radiocom, it's not really clear what the right abstraction is for your use case, without knowing what it is.  Generally, anything that has raw access to TCP sockets needs to do buffering of messages and parsing.  This is likely to stress the Microvium heap quite a bit.  Exposing higher-level network protocols requires a lot less work in the JavaScript, but more on the integration side.  For example, something like MQTT might be exposed as a simple callback for handling an event and a function for publishing to a specific node, so a small amount of JavaScript on the device can drive a lot of policy and not require you to reimplement an efficient C/C++ MQTT client library in interpreted JavaScript.