coder-mike/microvium #45: MVM_PORT_INT32_OVERFLOW_CHECKS clarification

TobyEalden, 2022-10-22:
Can you clarify [this comment about `MVM_PORT_INT32_OVERFLOW_CHECKS`](https://github.com/coder-mike/microvium/blob/main/native-vm/microvium_port_example.h#L83)?

It seems to imply that wrap-around will occur on integer arithmetic, but I'm seeing overflow behaviour on floats when this is set to zero (see output below).

Do I need to do something to force the variables to float, or is this what `MVM_PORT_INT32_OVERFLOW_CHECKS` is intended for? (To be clear if I set `MVM_PORT_INT32_OVERFLOW_CHECKS` to 1 I get the expected results).

The code I'm using is similar to:

```js
function fibonacci(commandId, payload) {
    let idx = 0;

    const doFormat = (val) => `{"idx": ${intToString(idx++)}, "fib": ${floatToString(val)}`;

    let first = 0.0;
    emitPayload(commandId, doFormat(first));

    let second = 1.0;
    emitPayload(commandId, doFormat(second));

    let fib = 0.0;
    const limit = stringToInt(payload);
    while (idx < limit) {
        fib = second + first;
        first = second;
        second = fib;
        emitPayload(commandId, doFormat(fib));
    }
}
```

Output is:

```
{"idx": 0, "fib": 0.000000
{"idx": 1, "fib": 1.000000
{"idx": 2, "fib": 1.000000
{"idx": 3, "fib": 2.000000
{"idx": 4, "fib": 3.000000
{"idx": 5, "fib": 5.000000
{"idx": 6, "fib": 8.000000
{"idx": 7, "fib": 13.000000
{"idx": 8, "fib": 21.000000
{"idx": 9, "fib": 34.000000
{"idx": 10, "fib": 55.000000
{"idx": 11, "fib": 89.000000
{"idx": 12, "fib": 144.000000
{"idx": 13, "fib": 233.000000
{"idx": 14, "fib": 377.000000
{"idx": 15, "fib": 610.000000
{"idx": 16, "fib": 987.000000
{"idx": 17, "fib": 1597.000000
{"idx": 18, "fib": 2584.000000
{"idx": 19, "fib": 4181.000000
{"idx": 20, "fib": 6765.000000
{"idx": 21, "fib": 10946.000000
{"idx": 22, "fib": 17711.000000
{"idx": 23, "fib": 28657.000000
{"idx": 24, "fib": 46368.000000
{"idx": 25, "fib": 75025.000000
{"idx": 26, "fib": 121393.000000
{"idx": 27, "fib": 196418.000000
{"idx": 28, "fib": 317811.000000
{"idx": 29, "fib": 514229.000000
{"idx": 30, "fib": 832040.000000
{"idx": 31, "fib": 1346269.000000
{"idx": 32, "fib": 2178309.000000
{"idx": 33, "fib": 3524578.000000
{"idx": 34, "fib": 5702887.000000
{"idx": 35, "fib": 9227465.000000
{"idx": 36, "fib": 14930352.000000
{"idx": 37, "fib": 24157817.000000
{"idx": 38, "fib": 39088169.000000
{"idx": 39, "fib": 63245986.000000
{"idx": 40, "fib": 102334155.000000
{"idx": 41, "fib": 165580141.000000
{"idx": 42, "fib": 267914296.000000
{"idx": 43, "fib": 433494437.000000
{"idx": 44, "fib": 701408733.000000
{"idx": 45, "fib": 1134903170.000000
{"idx": 46, "fib": 1836311903.000000
{"idx": 47, "fib": -1323752223.000000
{"idx": 48, "fib": 512559680.000000
{"idx": 49, "fib": -811192543.000000
{"idx": 50, "fib": -298632863.000000
{"idx": 51, "fib": -1109825406.000000
{"idx": 52, "fib": -1408458269.000000
{"idx": 53, "fib": 1776683621.000000
{"idx": 54, "fib": 368225352.000000
{"idx": 55, "fib": 2144908973.000000
{"idx": 56, "fib": -1781832971.000000
{"idx": 57, "fib": 363076002.000000
{"idx": 58, "fib": -1418756969.000000
{"idx": 59, "fib": -1055680967.000000
{"idx": 60, "fib": 1820529360.000000
{"idx": 61, "fib": 764848393.000000
{"idx": 62, "fib": -1709589543.000000
{"idx": 63, "fib": -944741150.000000
{"idx": 64, "fib": 1640636603.000000
{"idx": 65, "fib": 695895453.000000
{"idx": 66, "fib": -1958435240.000000
{"idx": 67, "fib": -1262539787.000000
{"idx": 68, "fib": 1073992269.000000
{"idx": 69, "fib": -188547518.000000
{"idx": 70, "fib": 885444751.000000
{"idx": 71, "fib": 696897233.000000
{"idx": 72, "fib": 1582341984.000000
{"idx": 73, "fib": -2015728079.000000
{"idx": 74, "fib": -433386095.000000
{"idx": 75, "fib": 1845853122.000000
{"idx": 76, "fib": 1412467027.000000
{"idx": 77, "fib": -1036647147.000000
{"idx": 78, "fib": 375819880.000000
{"idx": 79, "fib": -660827267.000000
{"idx": 80, "fib": -285007387.000000
{"idx": 81, "fib": -945834654.000000
{"idx": 82, "fib": -1230842041.000000
{"idx": 83, "fib": 2118290601.000000
{"idx": 84, "fib": 887448560.000000
{"idx": 85, "fib": -1289228135.000000
{"idx": 86, "fib": -401779575.000000
{"idx": 87, "fib": -1691007710.000000
{"idx": 88, "fib": -2092787285.000000
{"idx": 89, "fib": 511172301.000000
{"idx": 90, "fib": -1581614984.000000
{"idx": 91, "fib": -1070442683.000000
{"idx": 92, "fib": 1642909629.000000
{"idx": 93, "fib": 572466946.000000
{"idx": 94, "fib": -2079590721.000000
{"idx": 95, "fib": -1507123775.000000
{"idx": 96, "fib": 708252800.000000
{"idx": 97, "fib": -798870975.000000
{"idx": 98, "fib": -90618175.000000
{"idx": 99, "fib": -889489150.000000
```

TobyEalden, 2022-10-22:
Reproduction here https://github.com/TobyEalden/microvium-get-object/tree/double-overflow

coder-mike, 2022-10-22:
The intended behavior is that without `MVM_PORT_INT32_OVERFLOW_CHECKS `, numbers that can be encoded as int32 will have wrap-around behavior when using add, subtract, or multiply operations, as in [this test case](https://github.com/coder-mike/microvium/blob/0e292423e181ca01e785ffe3d41fc26e10f1b034/test/end-to-end/tests/number-operations.test.mvm.js#L43):

```js
 assertEqual(2_000_000_000 + 2_000_000_000, overflowChecks ? 4_000_000_000 : -294967296);
```

The intended use-case is if you're running on a particularly resource-constrained device and don't want the runtime overhead of the engine checking if every integer operation might overflow and need to be upgraded to a proper 64-bit float, and your script is working with numbers that you don't expect to overflow the signed 32-bit range.

Thanks for the code example -- I'll take a look.

coder-mike, 2022-10-22:
Yeah, actually, it looks like the result you're getting is expected. At index 47, it's adding `1_134_903_170 + 1_836_311_903`. The "correct" answer to this 2_971_215_073, but this is larger than a signed 32-bit integer, so it overflows to `-1_323_752_223`. If you want the engine to catch this overflow and instead use the proper floating-point arithmetic, `MVM_PORT_INT32_OVERFLOW_CHECKS` needs to be `1` to enable the checks.

Maybe I should clarify the documentation to explain what wrap-around behavior means. How about this:

```c
/**
 * Set to 1 to enable overflow checking for 32 bit integers in compliance with
 * ES262 standard.
 *
 * If set to 0, then operations on 32-bit signed integers have wrap-around
 * (overflow) behavior, like when adding 32-bit signed integers in C.
 *
 * Explanation: Microvium tries to use 32-bit integer arithmetic where possible,
 * because it's more efficient than the standard 64-bit floating point
 * operations. To give the appearance of 64-bit floating point, Microvium needs
 * to check when the result of such operations overflows the 32-bit range and
 * needs to be re-calculated using proper 64-bit floating point operations.
 * These overflow checks can be disabled to improve performance and reduce
 * engine size.
 *
 * Example: `2_000_000_000 + 2_000_000_000` will add to:
 *
 *   - `4_000_000_000` if `MVM_PORT_INT32_OVERFLOW_CHECKS` is `1`
 *   - `-294_967_296` if `MVM_PORT_INT32_OVERFLOW_CHECKS` is `0`
 *
 */
#define MVM_PORT_INT32_OVERFLOW_CHECKS 1
```

Also, I feel like the default should be to have these enabled, since they give compliant behavior. I'm not sure why I chose to disable the overflow checks in the example port file.

TobyEalden, 2022-10-24:
That's definitely an improvement. Arguably changing the name from `MVM_PORT_INT32_OVERFLOW_CHECKS` to `MVM_PORT_64_BIT_FLOATING_POINT` would be more explicit.

coder-mike, 2022-10-24:
The thing is, you can still do floating point operations with `MVM_PORT_INT32_OVERFLOW_CHECKS 0` because some numbers are already floating point. For example, `0.5 + 1` will always be `1.5`, even if `MVM_PORT_INT32_OVERFLOW_CHECKS` is false, because `MVM_PORT_INT32_OVERFLOW_CHECKS` only applies to overflows of the 32 bit integer range but `0.5` is not a 32-bit integer. Similarly  `8_000_000_000` is not a 32-bit integer, so `8_000_000_000 + 2_000_000_000` will produce `10_000_000_000` even if the 32-bit int overflow checks are off.

There is also already a `MVM_SUPPORT_FLOAT` flag. Microvium could theoretically have overflow checks without supporting floating point operations and vice versa. E.g. if you overflow without float support, it could just produce an error. Although it doesn't look like that's the actual behavior at the moment, it probably should be.

I could make things simple by just removing the flag and instead just always doing the check. Maybe the flag is just too confusing.

coder-mike, 2022-10-25:
Ok, I've removed it from the example port file so it doesn't confuse people, and because the checks are not that expensive anyway. Also, overflow of a signed int in C is technically undefined behavior, so even saying that it wraps is not technically accurate even if that's what it does on basically all real-world systems. But if you have it in your port file, it will still use it, otherwise it will fall back to a default of `1`.

closed 2022-10-25 by coder-mike