coder-mike/microvium #46: Register state mismatch assertion

TobyEalden, 2022-10-22:
I'm getting intermittent asserts failing here: https://github.com/coder-mike/microvium/blob/main/dist-c/microvium.c#L3544

I'm running essentially the [same code as here](https://github.com/TobyEalden/microvium-get-object), but integrated into another application and the `fibonacci` endpoint is being called multiple times periodically. I have tried to reproduce it in a minimal standalone example, but don't seem to be able to. It isn't easily reproducible, i.e. running the same sequence doesn't always cause the failure.

The only difference I can see between the `reg` and `regCopy` structs is the value of the `scope` field - does that narrow down the possible cause at all?

coder-mike, 2022-10-22:
"Intermittent asserts failing" - my favorite kind of bug 😆 

> The only difference I can see between the reg and regCopy structs is the value of the scope field - does that narrow down the possible cause at all?

Yes, that's helpful information.

Is it possible for you to tell me which host function is being called when it fails? (The value of the variable `hostFunction` near the assertion point).

I'm not sure yet, but I have one theory, which is that maybe the host function is directly or indirectly triggering a GC run, which is moving things around in memory, which may change the value of the `scope` register since it will point to the new location in memory after GC compaction. If that's the cause, then the behavior is fine but the assertion itself is wrong and should be changed/removed. I'll look into it and see what I can find.

TobyEalden, 2022-10-23:
I've managed to get it to a point where I can consistently reproduce the issue, i.e. it crashes every time in exactly the same place.

Unfortunately I can't reproduce it in the minimal example I created above, but essentially it's crashing in an equivalent host function to [this](https://github.com/TobyEalden/microvium-get-object/blob/double-overflow/main.js#L11), after 11 iterations of [this loop](https://github.com/TobyEalden/microvium-get-object/blob/double-overflow/main.js#L24).

Based on your theory I put a `printf("mvm_runGC\n");` [here](https://github.com/coder-mike/microvium/blob/main/dist-c/microvium.c#L5115), and sure enough it looks like the gc is run right before the crash:

[image]

Let me know if there's anything else I can try to help. FWIW if I set MVM_SAFE_MODE to 0 I can run many iterations and there is no crash.

coder-mike, 2022-10-23:
Ok thanks, that was helpful. It should be fixed now (in `main`), if you wouldn't mind checking.

For interest, I managed to reproduce it on my side with [the following test script](https://github.com/coder-mike/microvium/blob/HEAD/test/native/native-api.test.ts#L204-L240):

```js
const runGC = vmImport(1);

vmExport(1, runTest);

function runTest() {
  // Create some garbage to be collected
  [];

  // Create a closure. The scope for this closure will be allocated after
  // the above garbage, which means it will move during a GC cycle as the
  // heap is compacted.
  const closure = createClosure();

  // Run the closure, which calls the host to trigger a GC cycle. Since
  // the GC is triggered while the closure is active, the 'closure'
  // register will point to the closure, which moves. This tests that
  // nothing breaks if the current closure moves during a GC cycle during
  // a call to the host.
  closure();
}

function createClosure(x) {
  return () => {
    x; // Force this func to be a closure
    runGC();
  };
}
```

TobyEalden, 2022-10-24:
Great - thanks for this! I can confirm it has fixed the issue for me.

closed 2022-10-24 by TobyEalden