coder-mike/microvium #48: mvm_call() error on multiple declarations for same variable across for loops

Jerryxia32, 2022-11-22:
This reduced test case
```
console.log = vmImport(1);
function sayHello() {
	var arr = [1, 2, 3];
	for (var i = 0; i < 3; ++i) {
		console.log(arr[i]);
	}
	for (var i = 0; i < 3; ++i) {
		console.log(arr[i]);
	}
}
vmExport(1234, sayHello);
```
causes `mvm_call()` error with a return value of 12 when entering `sayHello()`, on tag 0.0.25.

My Javascript expertise is limited, but after some poking around I think it's the second `var i = 0` in the second for loop that chokes Microvium. If I just change the second `var i = 0` into `i = 0`, it works fine.

coder-mike, 2022-11-22:
Thanks for the reporting this. I'll look into it.

In the meantime, can I suggest using `let` instead of `var`. In JS, `var` declarations are hoisted to the function level, so two `var` declarations with the same name in the same function are actually the same function-level `var` declaration, which can lead to subtle bugs in your code. Here's some more detail about the differences [on Stack Overflow](https://stackoverflow.com/a/11444416/890587).

coder-mike, 2022-11-22:
I've fixed this on `main`. The fix will go out with the next release. But as I say, I do recommend using `let` (or `const`) instead of `var`.

commit 16dade28 (2022-11-22): Fix #48 bug with duplicate var declarations
commit a63294dd (2022-11-22): Merge pull request #49 from coder-mike/48-var-issue

Fix #48 bug with duplicate var declarations
closed 2022-11-22 by coder-mike