coder-mike/microvium #50: Microvium does not clean up stack memory if `mvm_free` is called from an FFI function

davidchisnall, 2022-11-24:
If `mvm_free` is called during FFI invocation, the stack region is not cleaned up by the `gc_freeGCMemory(vm)` call.  I am working around this locally by adding `vm_free(vm, vm->stack)` to `mvm_free`, but I don't know if that is the right fix.

Our code doing this still respects the 'you must not use `vm` after this call' requirement, we call this from an error handler that does not return from the FFI call.

coder-mike, 2022-11-24:
Funny story: I once tried to delete the `C:/Windows` folder on my windows computer (while booted into Windows). I was young and was planning on reinstalling anyway, and thought maybe the new installation would go faster if I helped it along by pre-deleting files. I expected it to fail at some point because Windows should be using those files, but it didn't - it carried on deleting itself (apparently) until the whole folder was gone. Then I realized in horror that I had accidentally deleted `D:/Windows` instead of `C:/Windows`. This machine was dual-booting 2 instances of Windows, and I deleted the wrong one. This was a family computer and the other instance of Windows my dad's. Oops!

Coming back to Microvium, what I understand is that you're trying to halt the VM, either by terminating the thread or by using `longjmp` or something, and need to clean up the memory. Since Microvium cleans up its own stack when it goes idle, but in this scenario it's not being allowed to return and unwind itself, the stack is still allocated and becomes a memory leak. Is that correct?

If so, yes, it should really cater to that scenario, so I'll modify `mvm_free` to also free the stack if it's allocated. And yes, your work-around sounds correct. Microvium has one allocation for the VM struct itself, one for the stack, and zero or more allocations for the heap as it grows. `mvm_free` currently frees all except the stack (`vm->stack`) so your solution should work.

coder-mike, 2022-11-24:
Looks like my recent commit automatically closed this ticket. But please let me know if this doesn't address your problem.

closed 2022-11-24 by coder-mike

davidchisnall, 2022-11-25:
Thanks.  I think your version generates bigger code than mine.  The `  if (vm->stack)` block is necessary only if `vm_free` doesn't handle null pointers.  I think this is the only place where you free anything that might be null.  Any embedding that's using a C-compatible `free` will, possibly we could have a hook to specify that an implementation does / doesn't handle null and only do the check afterwards?

I didn't need `vm->stack = 0;` because I put it after the `gc_freeGCMemory` call, so the zeroing is folded into the `memset` in paranoid mode, or unnecessary because the object is freed immediately after the store.

coder-mike, 2022-11-25:
Yeah, true, thanks.

commit 3366882b (2022-11-24): Fix #50 - mvm_free should also free the stack