coder-mike/microvium #56: Unsigned 32-bit arithmetic without floating point support?

davidchisnall, 2023-02-01:
We have a few places where we're passing a 64-bit value as two 32-bit integers in C (to avoid accidentally calling 64-bit division helpers and so on).  When we pass these into JavaScript, we need to pass them as signed 32-bit integers.  It would be nice if the Number representation could support 32-bit unsigned values, even if they hit a very slow path if they weren't representable as 32-bit signed values.

The main use case of this is a 64-bit time counter, where we need would like to be able to implement subtraction in JavaScript without a load of corner cases for the negative case.

coder-mike, 2023-02-02:
If I understand you correctly, you're looking for a number type in JavaScript which has unsigned 32-bit overflow semantics during arithmetic operations. Correct me if I'm wrong.

I don't see an easy way to add this to the engine itself. The type code nibble in the 16-bit allocation header is only 4 bits, giving 16 builtin types. One of these is `TC_REF_INT32` which is the signed 32-bit integer. To support a 32-bit unsigned type as well, I think we would need to reserve another of the 16 type codes for this. There are only 3 spare type codes at this stage, and they are all earmarked for particular future capability.

I would also consider the fact that in the ECMAScript spec, there are no overflow semantics at all for arithmetic operations. So if your JavaScript code depends on this behavior, then it's no longer truely JavaScript (e.g. if you run it in the browser or nodejs then it will have different behavior to running it on Microvium). The ability to disable float support and overflow checking was intended only for programs that are expected to never hit the overflow path in practice, so where the program still has ECMAScript compliant behavior in practice. (Maybe this is similar to the fact that the ECMAScript spec assumes infinite memory, but practical engines have finite memory and so the memory can overflow, but as long as your program doesn't _depend_ on the out-of-memory behavior then it still behaves according to the ECMAScript spec).

What about an alternative where you use an 8-byte `Uint8Array` instead, and implement the relevant operations in C?

davidchisnall, 2023-02-03:
> If I understand you correctly, you're looking for a number type in JavaScript which has unsigned 32-bit overflow semantics during arithmetic operations. Correct me if I'm wrong.

Not exactly, I'm looking for a way in JavaScript to have a `Number` that can hold any valid unsigned 32-bit value.  This is the case when floating-point is enabled (you can store any unsigned 52-bit value, if memory serves).  Unfortunately, when not in floating-point mode, I can construct integers from C only as signed 32-bit values, so I can't represent a quantity that doesn't fit in that range.

coder-mike, 2023-02-04:
Ok. Then my second point is invalid -- if your program never actually overflows an unsigned 32-bit integer, then it still may only rely on the ECMAScript-compliant part of Microvium and behave the same in Node etc.

After thinking about it, I have an idea that might work. Possibly you enable float support, but defining an `MVM_FLOAT64` as `int64_t` or `uint32_t` in the port file. This way, it doesn't need a new typecode since it will use `TC_REF_FLOAT64`. I also think that the overflow checking will just work the same way it already does. The checks for signed int32 overflow won't change. The checks for conversion back to int32 essentially won't change since they essentially just do a cast to `int32_t` and back to `MVM_FLOAT64` and see if the value is preserved (if it's preserved then it will save it as an `TC_REF_INT32`, otherwise it's an `TC_REF_FLOAT64`).

The changes required to the engine would be abstracting out macros for the floating point constants and primitives, such as `-0.0`, `isnan`, `isfinite`, and maybe `signbit`. And the `math.h` header, as you pointed out in a different ticket.

A potential gotcha would be that any floats in the snapshot will be encoded differently to what the runtime uses, since the snapshot should use a proper IEEE754 encoding.

Is there anything I'm missing or do you think this would work?

davidchisnall, 2023-02-04:
> After thinking about it, I have an idea that might work. Possibly you enable float support, but defining an MVM_FLOAT64 as int64_t or uint32_t in the port file

I believe Lua has a mode like this, with 64-bit as the big number type instead of `double`.  This moves quite a bit away from standard ECMAScript, but would be useful on devices without hard floating point support or enough instruction memory for a soft-float implementation.  

This would need us to include the 64-bit integer arithmetic support routines, but those are quite small.

coder-mike, 2023-02-05:
Ok, I've created a bunch of macros to abstract out the floating point implementation so you can define your own "floating point" implementation. As with any of the macros, you're free to provide a non-compliant implementation, at your own risk.

In particular, the following port macros might work but produce non-compliant float support (but I haven't tested it):

```c
// Support "floats" (in inverted commas)
#define MVM_SUPPORT_FLOAT

// Our custom float is actually just an unsigned 32-bit integer
#define MVM_FLOAT64 uint32_t

// Don't need math.h for this implementation
#define MVM_IMPORT_MATH 0

// Our custom float is never NaN
#define MVM_FLOAT_IS_NAN(x) 0
// Our custom float is never negative zero
#define MVM_FLOAT_IS_NEG_ZERO(x) 0
// Our custom float is always finite
#define MVM_FLOAT_IS_FINITE(x) 1
// If we use NaN in the JS code, it will show up as zero
#define MVM_FLOAT64_NAN 0
// If we use -0 in the JS code, it also shows up as zero
#define MVM_FLOAT_NEG_ZERO 0
```

As previously noted, floats in the snapshot, including those in the source code, and including any number value that does not fit in the _signed_ 32-bit integer range (e.g. `0xFFFFFFFF`), will be corrupted at runtime because they will be IEEE754 encoded in the ROM image but interpretted as `uint32_t` at runtime.

If you want to build a literal constant like `0xFFFFFFFF` in the source code (i.e. outside the signed 32-bit range), you'll need to  build it up as pieces, like `(0xFFFF * 0x10000) | 0xFFFF` (I haven't tested this expression, but I'm guessing it will work). Disclaimer: if Microvium has optimizers in the future, it's permissible for it to optimize this to the IEEE754-encoded float `0xFFFFFFFF` which will again be corrupted at runtime.

Be careful with the shift operators in this paradigm. An expression like `(0xFFFF << 16)`  is defined in ECMAScript to overflow using _signed_ 32-bit semantics, producing the number `-65536`. While the similar expression `0xFFFF * 0x10000` is defined in ECMAScript to use IEEE754 double-float semantics and produce the number `0xFFFF0000`. With the above macro definitions, I believe this will hold true (Microvium should detect that the latter overflows the `int32_t` and promote it to a "float" which is actually a `uint32_t`).

closed 2023-04-21 by coder-mike