coder-mike/microvium #59: Creating snapshots from JavaScript?

davidchisnall, 2023-03-14:
I am trying to automate some of the steps in a demo and have an Azure Function that does the compile step (JavaScript -> MVM bytecode) and pushes it to the device, but I don't see any documentation for this.  Did I miss something?  If there aren't any docs, can you point me to the code for the command-line tool?

coder-mike, 2023-03-14:
The getting-started guide has a section on [Making a Snapshot with the CLI](https://github.com/coder-mike/microvium/blob/main/doc/getting-started.md#making-a-snapshot-with-the-cli). The command is just:

```sh
microvium script.js
```

This runs the top-level code in the JS file, including `imports` and transitively all the top-level code in the import graph, and by-default produces a snapshot (unless you say `--no-snapshot`). 

The more detailed CLI documentation is available behind `microvium --help`

```
$ microvium --help
usage: microvium [-h] [-v] [-e "script"] [-s FILENAME] [--no-snapshot]
                 [--map-file FILENAME] [--output-disassembly] [--generate-lib]
                 [--generate-port] [--output-bytes] [--output-il]
                 [input [input ...]]

Microvium - A compact, embeddable scripting engine for microcontrollers for
executing small scripts written in a subset of JavaScript.

Positional arguments:
  input                 Input file to run

Optional arguments:
  -h, --help            Show this help message and exit.
  -v, --version         Show program's version number and exit.
  -e "script", --eval "script"
                        Evaluate the given script text and output snapshot
  -s FILENAME, --snapshot FILENAME
                        Snapshot filename to use for output
  --no-snapshot         Do not output a snapshot file
  --output-disassembly  Output disassembly of snapshot bytecode file
  --generate-lib        Interactively generate C runtime engine library
  --generate-port       Interactively generate microvium port file
                        (microvium_port.h)
  --output-bytes        Output bytecode as comma-separated hex, suitable for
                        use in a C constant
  --output-il           Output debug IL for each module
```

Is there anything else you'd like from these docs or that would help explain the process for other people?

davidchisnall, 2023-03-15:
Sorry, I realise my question was phrased ambiguously.  I meant using JavaScript code to create snapshots, not creating snapshots on the CLI from JavaScript source code.  The environment that I want to deploy the compiler into provides Node.js runners, but I can't run a child process.  I want to be able to write some JavaScript code that runs on v8, takes a string of JavaScript source code and generates the bytecode, using the npm package of Microvium.

The docs describe how to rehydrate a snapshot in JavaScript and in C, but only how to create the snapshot from the CLI.  I would like to create the snapshot in JavaScript on v8 and then rehydrate it in C with Microvium on the device.

coder-mike, 2023-03-16:
Ah, got it. No, there isn't documentation at the moment. But to get you going with it, you might do something like this:

```ts
import { Microvium, addDefaultGlobals, nodeStyleImporter } from 'microvium';

function compile(sourceText: string, filename: string) {
  // Create an empty VM
  const vm = Microvium.create();

  // Add Microvium's default globals such as `vmExport` and `vmImport`
  addDefaultGlobals(vm);

  // Create a function to resolve `import` statements in the script (optional)
  const importDependency = nodeStyleImporter(vm, {
    fileSystemAccess: 'unrestricted',
    allowNodeCoreModules: true,
    basedir: path.dirname(filename)
  });
  // Evaluate the script
  vm.evaluateModule({ sourceText, debugFilename: filename, importDependency });

  // Create the snapshot
  const snapshot = vm.createSnapshot();

  // If you want the disassembly for debug purposes
  const { disassembly } = Microvium.decodeSnapshot(snapshot);

  // The `snapshot.data` is a Node Buffer (Uint8Array) containing the snapshot bytes 
  return snapshot.data;
}
```

davidchisnall, 2023-03-20:
Thanks.  For no good reason, I was not using TypeScript, so I ended up with this:

```javascript
const Microvium = require('microvium');

function compile(sourceText) {
    // Create an empty VM
    const vm = Microvium.create();

    // Add Microvium's default globals such as `vmExport` and `vmImport`
    Microvium.addDefaultGlobals(vm);
    try {
        // Evaluate the script
        vm.evaluateModule({ sourceText, debugFilename: 'firmware.js' });

        // Create the snapshot
        const snapshot = vm.createSnapshot();

        // If you want the disassembly for debug purposes
        const { disassembly } = Microvium.decodeSnapshot(snapshot);
        // The `snapshot.data` is a Node Buffer (Uint8Array) containing the snapshot bytes
        return { "result": snapshot.data, "error": null }
    } catch (e) {
        return { "result": null, "error": e.message }
    }
}
```

This is sufficient for my Azure Function to take some JavaScript submitted from a web form, compile it, and send it to the device via the Azure IoT Hub.  I'll be presenting this as a demo at CyberUK in Belfast next month.

coder-mike, 2023-03-20:
Ok, great. Good luck for the presentation! If there's a recording of it, I'd be interested to see it.

closed 2023-04-21 by coder-mike