coder-mike/microvium #62: Casting string to integer/number

boogie, 2023-03-26:
Probably it is trivial, but I've found no documentation about it. If I have a string value, how can I convert it to number?

```
let str = '42';
let num1 = +str;
let num2 = Number(str);
let num3 = 0 + str;
let num4 = 1 * str;
```

I would say these are bugs, in normal JavaScript adding a string to a number should end up in a number.
These seems to be not working, and I think there's no parseInt method as well (however it's possible to implement).

coder-mike, 2023-03-26:
Hi. Yes, you're right. The code path for coercing a string to a number is currently not implemented. You're the first person to ask for this functionality. I'll leave this ticket open to track progress on it.

In the meantime, as you say, you can implement a method like parseInt in C and import it.

closed 2023-03-27 by coder-mike

boogie, 2023-03-28:
Thanks for adding the support. I don't need floats now.
I'm using both npm and source code. I've a JS => bytecode server with npm, and just added microvium C sources to my firmware code by downloading the source.