coder-mike/microvium #64: Plans for array operations?

boogie, 2023-03-30:
As far as I see, only push, length and [num] getters are currently supported for arrays. What is the plan about adding the basics like pop, shift, unshift? Maybe some more?

coder-mike, 2023-03-31:
The short answer is that there's a long-term plan to add them but no short-term plan, but you can add them yourself in your particular app if you need them.

The longer answer is this:

- prototype methods like `push` make the snapshot larger because the prototype is almost always reachable and so can't be garbage collected prior to taking the snapshot. This is true of most of the JS built-ins. 

- I have a proof-of-concept optimizer algorithm that is able to eliminate unused prototypes and prototype methods, among other things, but I don't have time to work on it at the moment to finish it off. My focus is currently on a WASM package of the Microvium runtime and then async/await.

- They could be added as optional libraries, so they don't add to the default snapshot size, but this just hasn't been my focus at the moment. As a rule of thumb, I'm focusing more on the syntactic features of the language, since users can add their own functions and methods as-needed but they can't add their own syntax. 

Having said that, if someone in the Microvium community wanted to create a library of these methods for other people to use, that would be great.

boogie, 2023-04-03:
I see that these make it larger, but hard to implement pop(), as it both returns the popped value and modifies the array. I'm happy to add them on my own and share the code, can you point me to how to do that? I would implement it in C, to not grow the bytecode significantly, but no idea how to add it to the object.

boogie, 2023-04-07:
Looks good and works well, thanks. :)

boogie, 2023-04-14:
Let me close this issue, I think it is already works well enough.

closed 2023-04-14 by boogie