coder-mike/microvium #67: string comparison

boogie, 2023-04-14:
I'm not sure if it's my implementation or an issue with the engine, but string comparison operators greater and smaller than are causing an abort.

```javascript
let x = '0';
let y = '9';
let result = x < y; // abort()
```

It seems that there's a test for it which is working properly, and I can confirm there's no issue with the JavaScript based engine, but it is not working with the C engine for me.

boogie, 2023-04-15:
I've checked the C implementation, and as I see the code is translated to VM_OP_NUM_OP bytecode, and only `+`, `===` and `!==` are type independent, but  `<`, `<=`, `>`, `>=`, `-`, `/`, `%`, `**`, and `*` are only working with numbers.

What is the plan related to this? It is not mentioned at the "Supported Language" doc page, so I was surprised. It can be OK, however I would expect a runtime error, not an abort. Abort cannot be catched and handled.

coder-mike, 2023-04-16:
Ah. Honestly, I didn't even think of that one. The operators like `**`, `*`, `/`, `%`, and `-` all coerce their operands to numbers. It didn't occur to me that inequality operators don't do this, but it's obvious in hindsight.

I'll add it to the to-do list and in the meantime I'll note it in the supported language doc page.