coder-mike/microvium #69: Running scripts directly in C

ReimuNotMoe, 2023-04-17:
Hi. First of all, thanks for making such an awesome project.

From my understanding, it's only possible to compile a script into bytecode in the nodejs runtime? And the C implementation can only execute the byte code? Is there a way to run scripts directly in the C implementation?

coder-mike, 2023-04-17:
> Hi. First of all, thanks for making such an awesome project.

Thank you! It's been a lot of work.

> From my understanding, it's only possible to compile a script into bytecode in the nodejs runtime? And the C implementation can only execute the byte code?

Yes and yes.

> Is there a way to run scripts directly in the C implementation?

No. It's quite a fundamental part of the design of Microvium that it doesn't support parsing at runtime. This is to cut down the size.

I've personally found that most use cases for a tiny JavaScript engine involve humans typing JavaScript into a larger computer and then the script being sent to the device. In these situations, it's normally feasible for the larger computer/device to pre-compile the script before sending it. 

But if that's something you need, take a look at the [Alternatives](https://github.com/coder-mike/microvium/blob/main/doc/alternatives.md) doc that lists some alternative engines. In particular, ELK, mJS, and XS might be good alternatives depending on your size constraints.

What's your use case?

ReimuNotMoe, 2023-04-24:
> But if that's something you need, take a look at the [Alternatives](https://github.com/coder-mike/microvium/blob/main/doc/alternatives.md) doc that lists some alternative engines. In particular, ELK, mJS, and XS might be good alternatives depending on your size constraints.

I chose Microvium particularly because I'm targeting 16bit MCUs that need weird attributes and other compiler magic to address a larger RAM/ROM, and Microvium has outstanding support of these.

> What's your use case?

Personally I just don't want to involve nodejs in the entire workflow. Also it would be really nice if you can edit the source js files on the MCU (like fine tuning stuff on a programmable industrial control panel, where it's usually inconvenient to use commercial grade PC nearby).

Thank you for your response. I'll give it a think.

closed 2023-11-07 by coder-mike