coder-mike/microvium #75: Deno JavaScript runtime

muhlpachr, 2023-07-30:
Thank you for great project.

Are there any plans to support Deno JavaScript runtime?

coder-mike, 2023-07-30:
Are you talking about running the compiler on Deno instead of node.js? I have no current plans to do this. What is your use case that requires this?

muhlpachr, 2023-07-30:
Yes, running compiler on Deno instead of node.js.

Use case is related to preparation of simple/small and resource limited bundle for JS MCU app developer to develop/prepare app snapshots.

coder-mike, 2023-07-30:
You can already do that with node.js, so what aspect of your situation benefits from Deno specifically?

muhlpachr, 2023-07-30:
I hope for better isolation with limits for memory and CPU resources without another os specific sandboxing.
I have no clue how much effort it takes to support Deno's runtime. It may not be worth doing that in the current Microvium state.

coder-mike, 2023-07-30:
Yeah, at the moment it would be quite difficult. The `microvium` package bundles both the compiler and a native build of the runtime engine which uses NAPI under the hood for native bindings. This could be replaced with the new WASM library but it would be a fair amount of work.

muhlpachr, 2023-07-30:
Thanks for info.

coder-mike, 2023-11-07:
After considering the effort involved, I've decided not to pursue Deno support for now, but I'll reconsider if there's more demand in the future. Thanks for your understanding.

closed 2023-11-07 by coder-mike