coder-mike/microvium #76: Creating a standard library

zap8600, 2023-08-24:
I would like to maje my own standard library. However, I want to use custom function names. How can I forward this before snapshotting?

coder-mike, 2023-08-24:
You can implement a library as an ES module, using `export` to make your library functions accessible to other modules. Or you can implement a library in C and expose those functions to JavaScript using `vmImport`. Do these work for you? If not, can you explain more about what you're trying to do?

zap8600, 2023-08-24:
I'm going to make it in C, so it has access to the hardware. I was going to make a `print` function, but microvium threw an error.
```~/microvium $ cat script.mvm.js
// script.mvm.js
print = vmImport(1);
function main() {
  print('Hello, World!', 75, 30);
}
vmExport(1234, main);
~/microvium $ microvium script.mvm.js
Microvium internal error (Unexpected state: Unit cannot be loaded because of missing required globals: print)
~/microvium $
```

coder-mike, 2023-08-24:
Just add a `const` in front of `print = ...` to make it into a module-level variable. Does that work for your use case?

zap8600, 2023-08-24:
> Just add a `const` in front of `print = ...` to make it into a module-level variable. Does that work for your use case?

That works, thank you. Quick question: Would it be possible to embed the Javascript compiler into a C application?

coder-mike, 2023-08-24:
No, that's not possible. It's just a trade off I made in the design that the compiler does a lot of the heavy lifting so that the runtime can be extra small, but that means it's not really feasible to run the compiler on the device unless your device is like a chunky Raspberry Pi running Linux and node.js.

zap8600, 2023-08-24:
> No, that's not possible. It's just a trade off I made in the design that the compiler does a lot of the heavy lifting so that the runtime can be extra small, but that means it's not really feasible [...]

Oh, OK. Thanks anyway. I like the project btw. I can't wait to see more.

coder-mike, 2023-08-24:
Great! Let me know if you have any other trouble with it.

closed 2023-08-24 by coder-mike