coder-mike/microvium #77: Importing functions with Multiple ways to call it

zap8600, 2023-08-29:
I'm trying to implement [`console.log`](https://developer.mozilla.org/en-US/docs/Web/API/console/log) in my standard library. However, there are different ways I can call it. How can I implement this?

coder-mike, 2023-08-29:
If you're talking about calling it at compile time and runtime, you can actually do this:

```js
// guest.js

vmExport(1, run);
console.log = vmImport(1, console.log);

console.log('console.log still works at compile time!')

function run() {
  console.log('console.log works at runtime too!')
}
```

The second argument to `vmImport` is like a "default" value that is used as a fallback if the current host doesn't have that numbered import.  The default compile-time host doesn't have any numbered imports available so it will use the default, which here is the compile-time `console.log` function.

Is that what you were asking?

zap8600, 2023-08-29:
I mean with arguments. For example, I can call it with multiple arguments or not, so the example for `console.log` provided in the native section of the getting started guide won't work due to `assert` and the lack of code to handle this. I'm wondering how I can handle a dynamic amount of arguments.

zap8600, 2023-08-30:
Nevermind. I think I have it figured out.

closed 2023-08-30 by zap8600

coder-mike, 2023-08-30:
Ok, I'm glad you figured it out. Do you want to post your solution here so others can see it if they come across a similar issue?

zap8600, 2023-08-30:
> Ok, I'm glad you figured it out. Do you want to post your solution here so others can see it if they come across a similar issue?

What I did is remove `assert`, go through each argument, get the type, and print it in the correct order.